MagicMirrorOrg/MagicMirror #2652: Suggestion: Globalized Date/Time

BKeyport, 2021-09-06:
I've been playing with the different date/time modules for JS - and noticed there's a concept mentioned in converting between them. We should make a standard API for when there is calls to an external script we rely on. Rather than directly using moment or luxon or what have you, you use: 

DateTime.<whatever> - and the system would automagically return the string as specified. This would make conversion from dead to live modules easy, we just reset the one file, and include the new scripts. 

This could apply to anything and everything we use. Request, moment, etc. Basically just put a shim between our work and their work, so modules will have less of a chance of breaking switching requires.

khassel, 2021-09-06:
As additional info: 

The project status of `moment` is described as "Moment.js is a legacy project, now in maintenance mode. In most cases, you should choose a different library."

See also discussion in [the forum](https://forum.magicmirror.builders/topic/15412/any-plan-to-replace-request-and-moment).

Alternatives are e.g. described [here](https://medium.com/swlh/best-moment-js-alternatives-5dfa6861a1eb).

rejas, 2021-09-08:
That idea sounds nice on paper and would help indeed when the library in question will have to be replaced. 

But I have alot of questions:

How could we encourage module developers to switch to the new API?

What calls are the modules using (with regards to moment js) How can we be sure to catch them all?

What happens to abandoned modules?

MichMich, 2021-09-14:
I'm all for replacing moment.js. Had a lot of issues with moment in the past, but indeed, it needs to remain a dependency for all 3rd party modules.

eouia, 2021-09-30:
https://github.com/MMRIZE/MMM-CustomElementTime

This is not for logic-related, but for displaying time, I will suggest this kind of global custom element to share the DateTime format.

closed 2022-04-27 by stale[bot]