MagicMirrorOrg/MagicMirror #2656: electron 14

khassel, 2021-09-09:
Electron 14 is released.

We should move to v14 **after** the October release and we have to wait until the corresponding spectron version v16 is released.

sdetweil, 2021-09-09:
why do u want to be that close to release cycle?
almost nothing good comes of it

khassel, 2021-09-09:
wrote already after next release so it would be in the January release and we have 3 months to test it on `develop`. I think we are totally contrary concerning update frequencies ...

MichMich, 2021-09-10:
3 months should be enough indeed.

khassel, 2021-09-29:
changed the title to v15, new electron main releases will come every 8 weeks, see [release blog](https://www.electronjs.org/blog/electron-15-0/).

khassel, 2021-11-26:
meanwhile electron v16 is out. The electron upgrade here depends on the [replacement of spectron](https://github.com/MichMich/MagicMirror/issues/2408#issuecomment-980452767).

khassel, 2021-11-28:
v16 is now on the `develop` branch.

In my docker setup on raspi 4 I get some errors concerning gpu on startup, but mm is running.

For getting rid of these messages there are 2 possibilities describes e.g. [here](https://jorgeavante.com/posts/running-an-electron-app-in-a-raspberry-pi-4/).

So before writing a PR adding `app.disableHardwareAcceleration();` to `js/electron.js` it would be good if someone else can test and confirm this, thanks!

Tested this already with a raspi 3 with electron v13 and v16 with no side effects.

rejas, 2021-11-29:
Mh... mm starts fine without that line 

After adding the line from the link right after app is declared, this line is in the logs:

0|MagicMir | [1048:1129/081539.472114:ERROR:sandbox_linux.cc(376)] InitializeSandbox() called with multiple threads in process gpu-process.

khassel, 2021-11-29:
yes, I'm aware of the sandbox error ...

But can you tell me more about your setup?
- pi version
- buster or bullseye
- node version
- electron v16?

Thanks.

rejas, 2021-12-01:
pi 4
bullseye "Linux mirrorpi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux"
node v16.13.0
electron v16 

:-)

rejas, 2021-12-01:
of course with latest apt-updates

khassel, 2021-12-01:
thanks, so this seems a problem of my docker setup and no general problem of mm ...

closed 2022-01-01 by khassel