MagicMirrorOrg/MagicMirror #2657: Control Magicmirror via numpad

Aruscha, 2021-09-09:
Hello, 
I would like to control my Magicmirror via numpad. Means: 
Number 1 > Page 1 
Number 2 > Page 2 

And so on and so forth. 
I already have the module Pages, only the change via Numpad does not work yet. 
How could I realize this?

luwol03, 2021-09-09:
You could listen to the keyboard events and send the page change command. You can either write such a Plugin your own or search  for an existing one. 

[MMM-KeyBindings](https://github.com/shbatm/MMM-KeyBindings) looks promising (haven’t tested it)

Aruscha, 2021-09-09:
What bothers me right now and what I don't understand is....

Pages wants you to work like this:
this.sendNotification("PAGE_CHANGED", 2);

But Keybinds has only Notification in the config...
Example
key: "ArrowLeft",
        state: "KEY_LONGPRESSED",
        notification: "CAROUSEL_PREVIOUS".

Is it so that I can simply enter "PAGE_CHANGED, 2" for Notification?

luwol03, 2021-09-09:
Yes, look at the payload option.

MichMich, 2021-09-10:
Please refer to http://forum.magicmirror.builders for support and troubleshooting. This doesn't seem to be a MagicMirror Core issue.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2021-09-10 by MichMich