MagicMirrorOrg/MagicMirror #2660: node V12 test is really needed?

eouia, 2021-09-10:
Node V12 is not `Active LTS`. It is on the `Maintenance LTS` stage.

Current dependency `Electron 13.2.3` is based on Node v14.17, Chromium 91, V8 9.1
And upcoming `Electron 14.0.0` will be based on Node v14.17, Chromium 93, V8 9.3

So the node V12 test is obsoleted at this moment, I think.

eouia, 2021-09-10:
I tried a new JS feature `string.replaceAll()` but it fails the node v12/ node v14 tests because it was introduced since node v15.
But it is implemented in the browser already(since Chromium 85, Edge 85, Firefox77, Opera 71, Safari 13.1), most parts of MM's code would be running on the browser. So I cannot catch up with the meaning of these server-sided browser emulating tests.

khassel, 2021-09-10:
> So the node V12 test is obsoleted at this moment, I think.

from my side the v12 tests could be removed, I never saw a different result between v12 and v14 as long as I look at them. End of life of v12 is 2022-04-30.

> So I cannot catch up with the meaning of these server-sided browser emulated tests.

there is already #2408 but I found no time so far to start with this.

rejas, 2021-09-10:
Thanks for your input and I understand your reasoning , alas:
- node12 being in Maintainance mode doesnt mean its not used anymore. It still gets security updates
- Electron might use node14 internally, but people can (and probably do) run their MM just in server mode and point any brwoser to it. 

As for your usecase: Would be easy to polyfill the replaceAll and add it to the code with a little note to remove it once node12 support is removed

eouia, 2021-09-10:
Yes, I can make a polyfill for `.replaceAll()`. And I'll do it soon.

BTW, for `node_helper` or other server-sided parts, v12 test has the worth. 

But to test browser(or electron)-sided parts, I still think that v12 test is meaningless.

 Of course, there could be cases of really old browsers like Midori or IE, but MM is supporting them still? (I'm not sure.) Especially`.replaceAll()` is already implemented in common modern browsers more than 1 year ago. Chromium 85 was released 2020-08-25, Firefox 77 was released 2020-06-01. The only node lately accepts this feature. The fact of what feature is valid in the real browsers cannot pass the server-based mimic test looks somewhat irony to me. ;)

closed 2021-09-10 by eouia