MagicMirrorOrg/MagicMirror #2663: local ics

Cobby67, 2021-09-16:
Is it possible to use an ics file local instead of a url link ?

sdetweil, 2021-09-16:
sure.. just put it in the calendar module folder

then use a url like this 
```
"http://localhost:8090/modules/default/calendar/????.ics" 
```
???? whatever the ics filename is

MichMich, 2021-09-16:
And just to prevent any confusion: the default port is 8080. 👍🏻

sdetweil, 2021-09-16:
yeh, should make that MM_port

Cobby67, 2021-09-16:
Wow!, thank you very much! I'll be able to see my prefered hockey team calendar (junior hockey)

sdetweil, 2021-09-16:
@Cobby67  problem? (re-open)

Cobby67, 2021-09-16:
I'm not a great programmer in javascript or nodejs... Can we see the date of the game, the team and the hour or it's just possible to see date and title?

Cobby67, 2021-09-16:
Ok, that will be allright!
Thank you!

Cobby67, 2021-09-16:
I can always import the ics in a mysql table and use the mmm_mysql... to see more column 
Have a nice day!

closed 2021-09-16 by Cobby67

sdetweil, 2021-09-16:
yep... lots of ways to solve specific issues