MagicMirrorOrg/MagicMirror #2666: Does not start in full screen

silchenko, 2021-09-19:
MM does not start in full screen mode. It only creates a window. I fixed it by manually editing electron.js and setting my native resolution while previously it was 800x600.

I used fresh pull install from today. This issue did not exist few months ago. Please fix.

MichMich, 2021-09-19:
Your issue is missing all the required information as stated in the issue template. Please fix. ;)

silchenko, 2021-09-19:
Platform: Raspberry Pi 3+
NPM: 6.14.15
MM version: 2.16.0
Description: MM window appears with 600x800 size in the left top corner of the window while I can still see the rest of the desktop
Configuration file: the default sample one

sdetweil, 2021-09-19:
did I have an electron app installed before?

delete the ~/.config/Electron folder

note the dot in front of config

you also could have set the size by using the electronOptions:{ height:1080, width:1920}
 in config.js

closed 2022-04-27 by stale[bot]