MagicMirrorOrg/MagicMirror #2671: MM 2.16.0 - Newsfeed not working more

The-Exterminator, 2021-10-01:
I get this error today, but i dont now why i get it, someone there can help. ???
the url work fine.
```
[01.10.2021 13:33.29.814] [ERROR] Newsfeed Error. Could not fetch newsfeed:  https://www.bt.dk/bt/seneste/rss FetchError: request to https://www.bt.dk/bt/seneste/rss failed, reason: certificate has expired
    at ClientRequest.<anonymous> (/home/pi/MagicMirror/node_modules/node-fetch/lib/index.js:1461:11)
    at ClientRequest.emit (events.js:315:20)
    at TLSSocket.socketErrorListener (_http_client.js:426:9)
    at TLSSocket.emit (events.js:315:20)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  type: 'system',
  errno: 'CERT_HAS_EXPIRED',
  code: 'CERT_HAS_EXPIRED'
}
```

The config.js code

```
    {
      module: 'newsfeed',
      position: 'bottom_bar',
      config: {
        feeds: [
          {
            title: 'BT',
            url: 'https://www.bt.dk/bt/seneste/rss'
          }
        ],
        showSourceTitle: true,
        showPublishDate: true,
        broadcastNewsFeeds: true,
        broadcastNewsUpdates: true
      }
    },
```

MichMich, 2021-10-01:
The ssl certificate of the feed you are using is expired. That's nothing on your side, but on theirs.

The-Exterminator, 2021-10-01:
Is still work on MM 2.15.0, but  not on 2.16.0

fpfuetsch, 2021-10-01:
Same issues here with 2.16.0 in the newsfeed module and the calendar module. Checked the certificates and they are still valid. Interestingly just specific certificates are marked as `expired`.

`
[01.10.2021 14:47.49.143] [ERROR] Newsfeed Error. Could not fetch newsfeed:  https://stadt-bremerhaven.de/feed FetchError: request to https://stadt-bremerhaven.de/feed failed, reason: certificate has expired
0|mm       |     at ClientRequest.<anonymous> (/home/pi/MagicMirror/node_modules/node-fetch/lib/index.js:1461:11)
0|mm       |     at ClientRequest.emit (events.js:315:20)
0|mm       |     at TLSSocket.socketErrorListener (_http_client.js:426:9)
0|mm       |     at TLSSocket.emit (events.js:315:20)
0|mm       |     at emitErrorNT (internal/streams/destroy.js:92:8)
0|mm       |     at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
0|mm       |     at processTicksAndRejections (internal/process/task_queues.js:84:21) {
0|mm       |   type: 'system',
0|mm       |   errno: 'CERT_HAS_EXPIRED',
0|mm       |   code: 'CERT_HAS_EXPIRED'
0|mm       | }
`

m1k6, 2021-10-01:
I guess this has to do with Let's Encrypt, as one of their CA's expired today.

More on their site: https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

eouia, 2021-10-01:
To fix; an upgrade node version is needed. at least v16(16.10 atm) will work.

fpfuetsch, 2021-10-01:
> I guess this has to do with Let's Encrypt, as one of their CA's expired today.
> 
> More on their site: https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

> "The newer root certificate (ISRG Root X1) is now widely trusted too" 
This is the root certificate of the "expired" certificates

rejas, 2021-10-01:
Can confirm that it happens even with valid certificates: Had my LetsEncrypt renewed in september for my calender, today the MirrorPi shows only an error.... Running my config locally (on windows) the calendar works without a problem

The-Exterminator, 2021-10-01:
> 
> 
> As I've written here; https://forum.magicmirror.builders/topic/15657/default-calendar-nextcloud-calendar-cert_has_expired/3?_=1633092214248
> 
> There could be some solutions to fix this;
> 
>     * upgrade node version as `lts`(v16) or  with `--use-openssl-ca` flag.
> 
>     * modify source code. Check `agentOptions.rejectUnauthorized` of `fetch` or `request` (whichever in source). I think that value or equivalent option should be false (to skip unauthorized result)
> 
>     * DANGER : process.env.NODE_TLS_REJECT_UNAUTHORIZED = 0 This is dirty and dangerous but quick solution. Dang... It is just a joke.
> 
> 
> I think the best solution will be upgrading node.

Get the samme error with the update you write :(

eouia, 2021-10-01:
Which method did you use?

The-Exterminator, 2021-10-01:
> 
> 
> Which method did you use?

request

bugsounet, 2021-10-01:
Hi, you have this again

const request = require('request').defaults({ rejectUnauthorized: false })

I use it temporaly

sdetweil, 2021-10-01:
newfeed moved to node-fetch in 2.15

```
const fetch = require("node-fetch");
```

bugsounet, 2021-10-01:
I know... but the result is the same error,
that why i re-use request "temporaly" (I have to add a new part! but time missing actually)

The-Exterminator, 2021-10-01:
> 
> 
> Hi, you have this again
> 
> transform:
> 
> ```js
> const request = require('request');
> ```
> 
> to:
> 
> ```js
> const request = require('request').defaults({ rejectUnauthorized: false });
> ```
> 
> I use it temporaly

wish file. ????

eouia, 2021-10-01:
I think just upgrade node to the latest `lts`, it will be solved.

The-Exterminator, 2021-10-01:
> 
> 
> I think just upgrade node to the latest `lts`, it will be solved.

Same error with 2.17.0 with both node 14 and 16

bugsounet, 2021-10-01:
> > Hi, you have this again
> > transform:
> > ```js
> > const request = require('request');
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > to:
> > ```js
> > const request = require('request').defaults({ rejectUnauthorized: false });
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I use it temporaly
> 
> wish file. ????

personal recoding

bugsounet, 2021-10-01:
@eouia : on Mac or SBC ?

rejas, 2021-10-01:
Upgrading my Pi4 to node 16 didnt solve the problem. Maybe a problem of the raspiOS?

rejas, 2021-10-01:
Could be an electron problem? Running MM in server mode and accessing it via brwoser works, running with electron doesnt...

rejas, 2021-10-01:
Looks like it is indeed,
see https://github.com/electron/electron/issues/31212 and https://community.letsencrypt.org/t/issues-with-electron-and-expired-root/160991/15

bugsounet, 2021-10-01:
don't know really
i'm writing a patch with node-fetch.
I just test it and works

do you want to see @rejas ?

rejas, 2021-10-01:
Thanks @bugsounet but could you test my PR if that works for you?

bugsounet, 2021-10-01:
Ah ... I will create a PR and you will more quick like me LOL

just 2 min, I will save may change and git reset

bugsounet, 2021-10-01:
Not tested with my Pi but in dev platform, it works

rejas, 2021-10-01:
Good to hear. So lets see how fast @MichMich can get this out :-)

MichMich, 2021-10-01:
In 3 months. 😂👍🏻

bugsounet, 2021-10-01:
[image]

bugsounet, 2021-10-01:
should i test in my pi before ?

MichMich, 2021-10-01:
Haha. Yes, I will. 
Hope to do it later tonight. 

A 2.17.1 PR to Master is welcome. 😬👍🏻

rejas, 2021-10-01:
PR with updated version number is ready

closed 2021-10-01 by MichMich

The-Exterminator, 2021-10-01:
Love when people help, Thanks to all.

The-Exterminator, 2021-10-05:
Work with 2.17.1

Thanks alot for the work :)

PR #2674 (2021-10-01): Update electron to fix certificate errors
Fixes #2671