MagicMirrorOrg/MagicMirror #2677: Clock Module... Look Ma, no hands!

SdeGeata, 2021-10-01:
Hello @MichMich;

I am running a new installation of MM on a Raspberry Pi4. The MM version is current.

My issue is the Clock module. I have it set for Analog Face and this worked fine in previous MM versions. However, in this latest install, there are no Hour or Minute hands, only the second hand. I have tried changing faces as well as commenting out any custom.css that I have to confirm that it is not related to that. It is not. I have also tried installing the version 2.18 dev version in case this was related to the previous issue with the faces not having any numbers. Also not the problem.

Any ideas?

[image]

rejas, 2021-10-02:
Hi @SdeGeata could you post your config of the clock module so I can take a look at it?

SdeGeata, 2021-10-02:
Hello @rejas Veeck!

Thanks for your prompt reply. Finally I wiped out all the Magic Mirror install and started fresh. The problem has been found. When I copied over my custom.css, I guess the main.css got copied over as well. The problem seems to have stemmed from that as there are so many changes in that css file.

Thanks for your help though. Problem solved.

closed 2021-10-02 by SdeGeata