MagicMirrorOrg/MagicMirror #2678: weather module: Feels Like temperature doesn't obey units correctly

kevinsivic, 2021-10-02:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi3

**Node Version**: v14.17.6

**MagicMirror Version**: 2.16.0

**Description**: 
When using the weather module with the 'current' type the feelsLike temperature is incorrect unless you explicitly set the tempUnits. It appears to use the Fahrenheit temperature but believes that it is a Celsius temperature and therefore tries to convert it to Fahrenheit. 

**Steps to Reproduce**: 
Configure the weather module similar to this:
```
{
	module: "weather",
	position: "top_left",
	config: {
		weatherProvider: "openweathermap",
		type: "current",
		units: "imperial",
		apiKey: "xxx",
	}
}
```

**Expected Results**: Feels like temperature displays correctly. 

**Actual Results**: Feels like temperature displays a number in the upper 1 or 200's

**Configuration**: See above

**Additional Notes**: 
This is fixed if the weather module is configured as follows:
```
{
	module: "weather",
	position: "top_left",
	config: {
		weatherProvider: "openweathermap",
		type: "current",
		units: "imperial",
		tempUnits: "imperial",
		windUnits: "imperial",
		apiKey: "xxx",
	}
}
```

kevinsivic, 2021-10-02:
I just noticed, this issue only occurs if the base 'units' config is set to something different than the weather modules 'units' configs.

Full steps to reproduce would include a config similar to this:

```
let config = {
	address: "0.0.0.0", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: [], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "weather",
			position: "top_left",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				units: "imperial",
				apiKey: "xxx",
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```

rejas, 2021-10-02:
Thanks for the bug report @kevinsivic 

Looking into it I noticed two things
a) the openweather provider didnt use the feels_like data from the API so it relied on the calculation provided by the weather module (feelsLike in weatherpovider). I provide a fix in the PR #2680

b) conceptually the units are a little weird: when nothing is specified in the weather config, it takes for all three (unit, wind, temp) the value from the general unit-config (for example "metric"):

https://github.com/MichMich/MagicMirror/blob/222a5f37791d9f8ae6182ef03487fdd509f1954f/modules/default/weather/weather.js#L15-L18

So if you change the `units` value in the weather config (to "imperial", the other two (temp/wind) still have the general config value (metric). This explains the behaviour you describe, since then the manual calucation mixes imperial and metric values and out comes something wrong :-)

Not sure how to improve that behaviour (if at all), will have to think about it.

In the meantime, can you check out my PR if that fixes your problem?

kevinsivic, 2021-10-04:
Thanks @rejas, that PR does fix the problem.

For (b), it may be enough to update the documentation to explicitly say that all of the units should be updated explicitly (or alternatively that you should use the general unit config).

rejas, 2021-10-05:
Glad to hear that it works.

Yes, maybe we need to just update the docs with more infos about that. Do you want to create your first PR for that (in the MagicMirror-Documentation repo)? https://github.com/MichMich/MagicMirror-Documentation/

PR #2680 (2021-10-05): Use feels_like data from openweathermap instead of calculating it
Fixes #2678

When getting the current weather from openweathermap, the provider ignored the feels_like data from the response, and instead calculated the information himself.

Due to how the units were configured, the result was way off (due to imperial and metric units being mixed together). This weird logic is not changed in this PR. That's for another day...