MagicMirrorOrg/MagicMirror #2707: [2759:1031/173943.629524:FATAL:electron_main_delegate.cc(253)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.

Ragziesoft, 2021-10-31:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the corresponding repository:

- karsten13/magicmirror: [https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)
- (deprecated) bastilimbach/docker-magicmirror: [https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 12 or later (recommended is 14).

**MagicMirror Version**: Please let us know which version of MagicMirror you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

Ragziesoft, 2021-10-31:
I am encountering the error message. 
[2759:1031/173943.629524:FATAL:electron_main_delegate.cc(253)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.

Please help

sdetweil, 2021-10-31:
run the fixsandbox script from my scripts , see 
https://github.com/sdetweil/MagicMirror_scripts

(altho you shouldnt be running MM as root)

Ragziesoft, 2021-10-31:
wow that was fast. Sorry but i am unable to follow how to do it(newbie). Could you please guide me. Thanks

Ragziesoft, 2021-10-31:
I copied script into a file fixsandbox.sh and executed it for home directory
sudo npm run fixsandbox.sh
i got this error
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/pi/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

I then stopped the pm2 which was running, and executed the fixsandbox script once again. Same error
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/pi/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-10-31T12_41_49_259Z-debug.log

sdetweil, 2021-10-31:
open a terminal window on MM system ( alt-spacebar, n to minimize MM) 

all the other scripts have a copy/paste

so, copy paste one of the others (into the terminal window), and then change the filename  from raspberry.sh  or whatever to fixsandbox

u can do this over ssh too

sdetweil, 2021-10-31:
>npm ERR! path /home/pi/package.json

you tried to npm start without being IN the MagicMirror folder

sdetweil, 2021-10-31:
so the command looks like this
```
bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/fixsadnbox)"
```

Ragziesoft, 2021-10-31:
Okay, so i executed the command and rebooted the pi. Still the same error

0|mm  | npm
0|mm  |  ERR! code ELIFECYCLE
0|mm  | npm ERR! errno 1
0|mm  | npm ERR!
0|mm  |  magicmirror@2.16.0 start: `./run-start.sh $1`
0|mm  | npm ERR! Exit status 1
0|mm  | npm ERR!
0|mm  | npm ERR! Failed at the magicmirror@2.16.0 start script.
0|mm  | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
0|mm  | npm ERR! A complete log of this run can be found in:
0|mm  | npm ERR!     /home/pi/.npm/_logs/2021-10-31T13_06_55_263Z-debug.log
PM2      | App [mm:0] exited with code [1] via signal [SIGINT]
PM2      | App [mm:0] starting in -fork mode-
PM2      | App [mm:0] online
1|mm     | > magicmirror@2.16.0 start /home/pi/MagicMirror
1|mm     | > ./run-start.sh $1
1|mm     | [3959:1031/183656.560431:FATAL:electron_main_delegate.cc(253)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
1|mm     | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP

sdetweil, 2021-10-31:
can you do from the MagicMirror folder 
```
ls  node_modules/electron/dist/chrome-sandbox -laF
```
and show the results
I get this 
```
base) sam@sams:~/MagicMirror$ ls node_modules/electron/dist/chrome-sandbox -laF
-rwsr-xr-x 1 root sam 50656 Oct 30 12:29 node_modules/electron/dist/chrome-sandbox*
(base) sam@sams:~/MagicMirror$ 
```
u don't need to reboot, just use the pm2 command to restart MagicMirror
```
pm2 restart MagicMirror
```

Ragziesoft, 2021-10-31:
okay i got this..
-rwsr-xr-x 1 root pi 1275900 Aug  6 09:49 node_modules/electron/dist/chrome-sandbox*

sdetweil, 2021-10-31:
ok, so it has the permissions

now, do 
logged on as user pi 
```
pm2 stop MagicMirror
cd ~/MagicMirror
npm start 
```

do u still get the error?

Ragziesoft, 2021-10-31:
Okay what i got is this

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.16.0 start /home/pi/MagicMirror
> ./run-start.sh $1

[31.10.2021 19:00.06.747] [LOG]   Starting MagicMirror: v2.16.0
[31.10.2021 19:00.06.756] [LOG]   Loading config ...
[31.10.2021 19:00.06.775] [LOG]   Loading module helpers ...
[31.10.2021 19:00.06.788] [LOG]   No helper found for module: MMM-pages.
[31.10.2021 19:00.06.791] [LOG]   No helper found for module: MMM-SystemStats.
[31.10.2021 19:00.06.793] [LOG]   No helper found for module: MMM-TuyaSL.
[31.10.2021 19:00.06.795] [LOG]   No helper found for module: MMM-HTTPRequestDisplay.
[31.10.2021 19:00.08.092] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.094] [LOG]   Module helper loaded: MMM-json-feed
[31.10.2021 19:00.08.096] [LOG]   No helper found for module: internet-monitor.
[31.10.2021 19:00.08.106] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.108] [LOG]   Module helper loaded: MMM-MicrosoftToDo
[31.10.2021 19:00.08.109] [LOG]   No helper found for module: MMM-Sounds.
[31.10.2021 19:00.08.155] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.157] [LOG]   Module helper loaded: MMM-Bose-Soundtouch
[31.10.2021 19:00.08.171] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.172] [LOG]   Module helper loaded: MMM-MWWordOfTheDay
[31.10.2021 19:00.08.182] [LOG]   No helper found for module: MMM-FlipClock.
[31.10.2021 19:00.08.192] [LOG]   No helper found for module: MagicMirror-QuoteCatalog.
[31.10.2021 19:00.08.194] [LOG]   No helper found for module: MMM-Admin-Interface.
[31.10.2021 19:00.08.195] [LOG]   No helper found for module: MMM-network-signal.
[31.10.2021 19:00.08.197] [LOG]   No helper found for module: mmm-systemtemperature.
[31.10.2021 19:00.08.219] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.220] [LOG]   Check MagicMirror version for node helper 'MMM-LocalTemperature' - Minimum version: 2.2.1 - Current version: 2.16.0
[31.10.2021 19:00.08.222] [LOG]   Version is ok!
[31.10.2021 19:00.08.223] [LOG]   Module helper loaded: MMM-LocalTemperature
[31.10.2021 19:00.08.226] [LOG]   No helper found for module: alert.
[31.10.2021 19:00.08.616] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.617] [LOG]   Module helper loaded: updatenotification
[31.10.2021 19:00.08.963] [LOG]   Initializing new module helper ...
[31.10.2021 19:00.08.965] [LOG]   Module helper loaded: calendar
[31.10.2021 19:00.08.968] [LOG]   No helper found for module: compliments.
[31.10.2021 19:00.08.974] [LOG]   
 and then after many more no helper found.... she starts
 what do i do next...i am scared to reboot...

sdetweil, 2021-10-31:
run the fixuppm2 script from my scripts.
down at the bottom of the page

Ragziesoft, 2021-10-31:
Okay so i did that and restarted pm2 restart mm.sh it says
[PM2][ERROR] Permission denied, to give access to current user:
$ sudo chown pi:pi /home/pi/.pm2/rpc.sock /home/pi/.pm2/pub.sock

i then execute the command and restart pm2, this time it restarts but MagicMirror does not start,,,  back to square one...

sdetweil, 2021-10-31:
>pm2 restart mm.sh

nooooooo... 

pm2 status
shows the defined apps..

my script sets up MagicMirror as the name 

so, pm2 start MagicMirror or pm2 start 0 

fixup knows what to do.. 

looks like you did something with sudo .. always a mess

sdetweil, 2021-10-31:
i would start over.. flash the card again 
and then run my script.. it does everything, just copy/paste that one line

Ragziesoft, 2021-10-31:
oops......
i spent many many hours installing and configuring the modules which includes PIR sensor and temp and humidity sensor modules... ( they take some complicated setups) its gonna take me ages to reconfigure everything

sdetweil, 2021-10-31:
ok, so, i would uninstall pm2, and then u my fixup script again 

```
pm2 kill
npm remove pm2 -g
reboot
```

make sure to remove the ~/.pm2  folder
```
from home/pi 
sudo rm -rf .pm2
```
then run my fixup script

Ragziesoft, 2021-10-31:
okay followed instructions to the point. so it started working. I was happy, then i rebooted....
and we are back to where we started...
pi@raspberrypi:~ $ pm2 status
[PM2][ERROR] Permission denied, to give access to current user:
$ sudo chown pi:pi /home/pi/.pm2/rpc.sock /home/pi/.pm2/pub.sock
pi@raspberrypi:~ $ pm2 start MagicMirror
[PM2][ERROR] Permission denied, to give access to current user:
$ sudo chown pi:pi /home/pi/.pm2/rpc.sock /home/pi/.pm2/pub.sock
so i once again execute the command...and the process starts, but the 
MagicMirror does not...
[image]

sdetweil, 2021-10-31:
see the last post here 
https://stackoverflow.com/questions/49563533/pm2-startup-script-for-non-root-user-permission-denied

this will list out the systemctl services for pm2 

sudo systemctl -a | grep pm2

Ragziesoft, 2021-10-31:
pi@raspberrypi:~ $ sudo systemctl -a | grep pm2
● pm2-pi.service                                                                               loaded    failed   failed    PM2 process manager
  pm2-root.service                                                                             loaded    active   running   PM2 process manage

sdetweil, 2021-10-31:
ok,  lets look in the log file 

sudo nano /var/log/daemon.log
ctrl-w, pm2

see this topic 
https://github.com/Unitech/pm2/issues/2747

need to find out while it failed.. if timeout then easy change.. if not...

Ragziesoft, 2021-10-31:
okay... timeout entr
[image]
y

sdetweil, 2021-10-31:
yeh, so increase its timeout by editing the service file as described in the stackoverflow post 

the pm2-pi service failed  cause the process failed too many times in a row (quickly)

Ragziesoft, 2021-10-31:
[Service]
Type=forking
User=pi
LimitNOFILE=infinity
LimitNPROC=infinity
LimitCORE=infinity
Environment=PATH=$PATH:/usr/bin:/bin:/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Environment=PM2_HOME=/home/pi/.pm2
PIDFile=/home/pi/.pm2/pm2.pid
Restart=on-failure

does the last line have to be
RestartSec=30 ??

sdetweil, 2021-10-31:
i assume u add it at the end of that section

sdetweil, 2021-10-31:
i do not know.. pm2 is not my code.. no one here has had this problem before.. that I know of in 4 years now

sdetweil, 2021-10-31:
i would be another sd card, install mm first, and migrate your stuff over...

Ragziesoft, 2021-10-31:
I just don't know how it broke. Everything was working fine, until i changed my monitor and then it crashed, Initially i thought the monitor was an issue so i reverted back the older monitor, but it wouldn't work there too. so now what...
Based on your help and advice when i change to MagicMirror directory and run npm start, it works fine  until i do ctrl c, then it just stops. SO npm start works but PM2 does'nt. What cud be the issue.

sdetweil, 2021-10-31:
its a pm2 issue.. 

pm2 keeps the process running

npm start does not

Ragziesoft, 2021-11-02:
So i did a reinstall of MM by the manual method , and the default config file fired up beautifully and MM was back. Then i painfully started reinstalling one module after another. Each time i checked to see if everything was fine by restating pm2. However, it crashed again, I then went back to the last module installed and deleted it, but the MM would not display anything. Now the most interesting part.... I then accidently discovered that if i use the basic config.js file and restart MM it starts working. I then switched to the new config.js file (with additional modules installed), while the mirror is still running running and once i refresh the browser, the new config file fires up with all the modules installed so far running fine (except Google Photos,, which has not been running so far,,,but that is another issue maybe). 
But now when i run pm2 restart, the MM crashes once again and MM goes blank. I repeated the process of using basic config.js file first and switch to new config.js file while the mirror is running many times and each time the same thing happens.
What is wrong, any help will be greatly appreciated. I have spent two sleepless nights already with no clue so far. It is the same issue we were trying to resolve these last few days.
Please help

sdetweil, 2021-11-02:
can u tell me the modules, and walk thru the process of adding them, the commands u issue and the folder you are in at the time

refreshing MM after modifying config.js gets the browser side loaded, but the helpers are not..   so not really loaded..

Ragziesoft, 2021-11-02:
[image]
i did a git clone of all modules. Then updated the config.js file each time. The folder is /home/pi/MagicMirror

sdetweil, 2021-11-02:
no.. something is damaging the files.. SO, I need to know   EXACTLY which folder you are in for every operation

after  installing MM..  for every module and which module 

u git clont every module
do you npm install for any module with a package.json file??  you must (in such module folder)

Ragziesoft, 2021-11-02:
Okay so after i chose  a module from the github, i copy the link of the code then i move to /MagicMirror/modules folder and run the script. Then i move to the specific module folder and do an npm install. Then i move to config folder and update the config.js file as required for the module.

sdetweil, 2021-11-02:
and then do a pm2 restart MagicMirror

and after which module does it break?

Ragziesoft, 2021-11-02:
Yes i do a pm2 restart mm after every module installed.  Well i guess when i was installing google photos in one instance so i deleted it, then once while installing MMM-Json-feed. But when ever i had this issue i deleted the module and commented out the config.js file. And did a pm2 resatart but it would not restore the screen display.
Even now i re installed google photos and did the configuration as explained in the github page including generating new token. The screen shows loading.... (after i tricked it with basic config file and switching to new config file as explained above)

sdetweil, 2021-11-02:
well, there is MODULE not working and there is MM not working

here is about MM not working

every module is different, some detect and recover from errors, some report them, some do not. 
Loading... is the sample default content displayed by modules until their data arrives. 
if it doesn't arrive (bad key, bad config, bad api,  ???  then 

two places to debug

in the window where u start MM, (npm start), if you use pm2, then pm2 logs --lines=???
I always use npm start when adding new modules so I can capture the output easier

then other debug is in the browser 
ctrl-shift-i, select the console tab, and scroll up
u can filter by module, by putting a unique part of the module name in the filter field  (goog) for googlephotos for example

Ragziesoft, 2021-11-02:
Ok, thats helpful...let me once again go through the grind... i am not giving up till i get some success...
thanks for being so prompt Sam, its very very encouraging......i will keep posting progress

Ragziesoft, 2021-11-03:
Finally.... my MM issue has been solved. Yippee. After painstakingly going through every setting up procedure, i finally managed to solve the issue. Things i found out was,
1. There was a lot of dead code(configuration entries of previously installed but removed modules) in the config.js file. I cleaned it up.
2. There were two modules that were broken and weren't loading(MMM-MWWordoftheday, and MMM-Wordoftheday) causing the MM to go into a loop. I deleted them. The log file was showing the issue with the modules.
3. Google photos finally is up and running. I reinstalled everything and also had to regenerate the token.
4. One module which is still not working is the Local temperature module which is connected to the DHT-22 sensor. Though it is configured and installed correctly(I hope) I still need to check out if the sensor is working(It was until a few days back before my MM broke). However, no error is shown in the log.
I big thank you to Sam for being there and guiding me through the process. I learnt quite a bit in the process. Highly appreciated sir.
Thanks

closed 2021-11-03 by Ragziesoft

sdetweil, 2021-11-03:
the two modules word of the day  is this 

https://forum.magicmirror.builders/topic/15778/fix-for-black-screen-in-2-16-and-later/1