MagicMirrorOrg/MagicMirror #2710: 3rd Party Modules not Appearing

bzoit, 2021-11-16:
Hello everyone! I'm trying to add the MMM-Jast and the MMM-cryptocurrency modules to my mirror but they either partially or don't appear at all. The cryptocurrency is just showing the bar at the top but no logo or price, and the jast module isn't appearing at all. My config.js is correct so I'm stumped. If anyone could help, that'd be very appreciated.

Config.js: pastebin.com/MPbD9kpW

[image]

sdetweil, 2021-11-16:
you should post issues to the module GitHub site, for the module author to address.

bzoit, 2021-11-16:
> you should post issues to the module GitHub site, for the module author to address.
> 
> 
> 
> and possibly post in the forum

Sorry, I'll do that.

closed 2022-04-27 by stale[bot]