MagicMirrorOrg/MagicMirror #2711: npm trhow error, cannot run MagicMirror

HecTicMoose, 2021-11-16:
Hey y'all, I've been having a couple issues after having my MagicMirror2 for a while. Main issue is I keep getting this throw error when I try to check my version of npm( npm -v ). 

```
npm -v
node:internal/modules/cjs/loader:936
  throw err;
  ^

Error: Cannot find module 'semver'
Require stack:
- /usr/share/npm/lib/utils/unsupported.js
- /usr/share/npm/bin/npm-cli.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/share/npm/lib/utils/unsupported.js:2:14)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/usr/share/npm/lib/utils/unsupported.js',
    '/usr/share/npm/bin/npm-cli.js'
  ]
}
```

This is all the code that pops up and any help would make me very thankful. 

```
node -v

v10.24.0

nodejs -v
v10.24.0
```

I've attempted to update both node and nodejs along with npm but nothing seems to work and i'm not entirely sure what the problem is. Thank you for any help!

sdetweil, 2021-11-16:
try this 
```
cd ~/MagicMirror
rm -rf node_modules
npm install
```
if that works, watch out for this 
https://forum.magicmirror.builders/topic/15778/fix-for-black-screen-in-2-16-and-later

but MM version 2.17 needs node 14. 

do this to upgrade node
```
cd ~
curl -sL https://deb.nodesource.com/setup_14.x | sudo -E bash -
sudo apt-get install -y nodejs
sudo apt-get install -y npm
```

OR, run my MM upgrade script  (which does all this AND upgrades MM to latest  and fixed the black screen issues) 
see https://github.com/sdetweil/MagicMirror_scripts

HecTicMoose, 2021-11-16:
I did everything you said to do and it ended up fixing the throw error, but like you said it went to black screen. I tried running it with the default config file and it works fine so it seems to only be a problem with my config file. These are some of the errors it comes up with for some modules.

```
[16.11.2021 18:12.42.523] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v11.5-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
```

it also says this:

`code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[16.11.2021 18:12.42.553] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[16.11.2021 18:12.42.553] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
`

any tips?

sdetweil, 2021-11-16:
Google assistant, I think u have to run 
```
npm run upgrade 
```
in it's module folder

HecTicMoose, 2021-11-17:
After navigating to the modules folder for each of the modules I have installed and running the `npm run upgrade` code this similar error message pops up for each.

```
npm ERR! Missing script: "upgrade"
npm ERR! 
npm ERR! To see a list of scripts, run:
npm ERR!   npm run

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-11-17T04_24_39_128Z-debug.log
```
I'm really not entirely sure what happened to my mirror because one day it just up and stopped working. I believe the only problem now is that all my modules need to be updated to the proper versions. Does that mean that I will have to reinstall all of my modules and reconfigure them all to how they were oringially? So sorry for any inconvience but thank you so much for the help!

sdetweil, 2021-11-17:
run my upgrade script, see
https://github.com/sdetweil/MagicMirror_scripts

it does not update module source

you do that by git pull
in each module folder. but many modules have not been updated

if u already did git pull for mm, then my upgrade needs force as a parm instead of apply

HecTicMoose, 2021-11-17:
I tried running your upgrade script and it seemed to upgrade MM perfectly without any problems however, it still goes right to black screen even after running everything you told me. The default config file works perfectly it genuinely just seems to be a problem with either one of multiple of my modules. Thank you for all of your help!

sdetweil, 2021-11-17:
look at the messages from where u start mm.

either npm start
or 
pm2

if pm2 , then pm2 logs --lines=50 will print out 50 lines

sdetweil, 2021-11-17:
my latest upgrade, SHOULD have handled the black screen

can u post the~/MagicMirror/installers/upgrade.log here as an attachment