MagicMirrorOrg/MagicMirror #2713: clock module shows 1 week wrong. think it uses calendar for 2020

paarak, 2021-11-18:
"version": "2.17.1",

[image]

rejas, 2021-11-18:
Hi, could you please post your clock entry from the config.js so we can take a look?

MichelBarneveld, 2021-11-22:
Change the language in your config.js to "en-gb" (British English) if you want to have the interface in English and you are in europe, instead of "en" (United States English). It will show the correct week. 

That is because there is an ISO standard for the first week of the year and some countries like the US not following that standard ;-) 
Moment.js uses the locale information to determine which method to use for weeknumbers.

For most countries the first week of the year is the first week with 4 days in the new year, like in Europe.
However there are some countries that uses the first week that has the first day of the year in it, like the US.

So this year the first day of the year was on a friday. Since Friday, Saturday and Sunday are only 3 days, the first week of this year started on Monday 4th of January in most countries. However in the US that was already week 2, since the first day of the year was the week before on Friday. So the US is 1 week ahead on Europe this year.

See also: https://en.wikipedia.org/wiki/ISO_week_date#First_week
and: https://en.wikipedia.org/wiki/ISO_week_date#Other_week_numbering_systems

paarak, 2021-11-28:
[image]

paarak, 2021-11-29:
> Hi, could you please post your clock entry from the config.js so we can take a look?

[image]

closed 2022-04-27 by stale[bot]