MagicMirrorOrg/MagicMirror #2714: weather module for ukmetoffice showing cors error

sdetweil, 2021-11-26:
Access to XMLHttpRequest at 'http://datapoint.metoffice.gov.uk/public/data/val/wxfcs/all/json/94004601?res=3hourly&key=fribble' from origin 'http://0.0.0.0:8090' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.

Access to XMLHttpRequest at 'http://datapoint.metoffice.gov.uk/public/data/val/wxfcs/all/json/94004601?res=3hourly&key=fribble' from origin 'http://localhost:8090' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
```
		{
			module: "weather",
			position: "top_right",
			config: {
        weatherProvider: "ukmetoffice",
        apiBase :'http://thingproxy.freeboard.io/fetch/http://datapoint.metoffice.gov.uk/public/data/val/wxfcs/all/json/',
        locationID: "94004601",
        apiKey: "fribble"
         }
}
```
should fail with unauthorized   as fribble is a bad apikey

this is the generated url request 
```
http://datapoint.metoffice.gov.uk/public/data/val/wxfcs/all/json/94004601?res=3hourly&key=fribble
```
2.17,1

khassel, 2021-11-26:
The used cors-provider in darksky provider `https://cors-anywhere.herokuapp.com/` seems to work (instead of `http://thingproxy.freeboard.io/fetch/`).

Getting now expected 401 (have no key).

sdetweil, 2021-11-27:
when I do this 
```
https://cors-anywhere.herokuapp.com/http://datapoint.metoffice.gov.uk/public/data/val/wxfcs/all/json/94004601?res=3hourly&key=fribble
```
I get a a nothing here yet page 
[image]

jptrsn, 2022-01-12:
I'm seeing a similar cors error using the envcanada weather provider.

Config.js entry:
```
{
    module: "weather",
    position: "top_left",
    config: {
        weatherProvider: "envcanada",
        siteCode: 's0000098',
        provCode: "QC",
        type: "current",
        showPeriod: false,
        showHumidity: true,
        location: "Salaberry-de-Valleyfield"
    }
},
```

console error:
> Access to XMLHttpRequest at 'https://thingproxy.freeboard.io/fetch/https://dd.weather.gc.ca/citypage_weather/xml/QC/s0000098_e.xml' from origin 'http://192.168.x.x:xxxx' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.

khassel, 2022-01-13:
I can confirm this error. So using `https://thingproxy.freeboard.io/fetch/` as cors proxy seems not to work anymore. It is difficult to find a working one, the above mentioned `https://cors-anywhere.herokuapp.com/` now wants a button click in the web ui before you can use it ...

khassel, 2022-01-28:
@jptrsn the above PR is merged so you could test if `envcanda` works now. You have to use the `develop` branch for testing this (next official release is 1.April). I would recommend to clean up (`rm -rf node_modules`) and run a new `npm install --only=prod`.

@sdetweil while testing this I had no problems with `ukmetoffice` when running without a cors proxy. If you still have cors problems with this provider you could also test the `develop` branch and add `useCorsProxy: true,` in the config:

```js
                {
                        module: "weather",
                        position: "top_right",
                        config: {
                                weatherProvider: "ukmetoffice",
                                useCorsProxy: true,
                                locationID: "322942",
                                apiKey: "xxx"
                        }
                },
```

Let me know if it works.

sdetweil, 2022-01-28:
ok, wasn't me directly. will advise the user

jptrsn, 2022-02-18:
I'm using a docker-based approach, so I will wait on the next production release and rebuild the image. Currently don't have any dedicated hardware for this.

sdetweil, 2022-06-29:
this doesn't work for ukmetofficehub.. doesnt use the provider fetchData... garf..

khassel, 2022-06-29:
tested `ukmetofficehub` without any problems

sdetweil, 2022-06-29:
I just tested on 2.20, develop, and it doesn't call providers/fetchData..  so can't get cors fix.

send me email, I can provide failing apikey info
same userid at gmail

khassel, 2022-10-02:
@sdetweil can this be closed?

closed 2022-10-02 by sdetweil