MagicMirrorOrg/MagicMirror #2717: UnhandledPromiseRejectionWarning on launch

josselynL, 2021-11-29:
**Platform**: RPi 3 on Raspbian
Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: 16

**MagicMirror Version**: 2.17.1

**Description**: Describe here https://forum.magicmirror.builders/topic/15933/unhandledpromiserejectionwarning-on-launch
UnhandledPromiseRejectionWarning on launch.
No problem when updatenotification is disabled.

```
pi@MagicMirror:~/MagicMirror $ npm run start

> magicmirror@2.17.1 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[29.11.2021 13:35.32.237] [LOG]   Starting MagicMirror: v2.17.1
[29.11.2021 13:35.32.250] [LOG]   Loading config ...
[29.11.2021 13:35.32.263] [LOG]   Loading module helpers ...
[29.11.2021 13:35.32.267] [LOG]   No helper found for module: alert.
[29.11.2021 13:35.32.301] [LOG]   Initializing new module helper ...
[29.11.2021 13:35.32.303] [LOG]   Module helper loaded: updatenotification
[29.11.2021 13:35.32.306] [LOG]   No helper found for module: clock.
[29.11.2021 13:35.32.622] [LOG]   Initializing new module helper ...
[29.11.2021 13:35.32.624] [LOG]   Module helper loaded: calendar
[29.11.2021 13:35.32.626] [LOG]   No helper found for module: weather.
[29.11.2021 13:35.32.629] [LOG]   No helper found for module: compliments.
[29.11.2021 13:35.32.631] [LOG]   No helper found for module: MMM-Traffic.
[29.11.2021 13:35.32.673] [LOG]   Initializing new module helper ...
[29.11.2021 13:35.32.675] [LOG]   Module helper loaded: newsfeed
[29.11.2021 13:35.32.679] [LOG]   Initializing new module helper ...
[29.11.2021 13:35.32.681] [LOG]   Module helper loaded: MMM-Jeedom
[29.11.2021 13:35.32.683] [LOG]   All module helpers loaded.
[29.11.2021 13:35.32.846] [LOG]   Starting server on port 8080 ...
[29.11.2021 13:35.32.872] [LOG]   Server started ...
[29.11.2021 13:35.32.875] [LOG]   Connecting socket for: updatenotification
[29.11.2021 13:35.32.878] [LOG]   Connecting socket for: calendar
[29.11.2021 13:35.32.880] [LOG]   Starting node helper for: calendar
[29.11.2021 13:35.32.881] [LOG]   Connecting socket for: newsfeed
[29.11.2021 13:35.32.883] [LOG]   Starting node helper for: newsfeed
[29.11.2021 13:35.32.885] [LOG]   Connecting socket for: MMM-Jeedom
[29.11.2021 13:35.32.886] [LOG]   Sockets connected & modules started ...
[29.11.2021 13:35.33.271] [LOG]   Launching application.
[29.11.2021 13:35.37.108] [LOG]   Create new calendarfetcher for url: http://p28-caldav.icloud.com/published/2/XXX - Interval: 300000
[29.11.2021 13:35.37.170] [LOG]   Create new calendarfetcher for url: http://p28-caldav.icloud.com/published/2/XXX - Interval: 300000
[29.11.2021 13:35.37.420] [LOG]   Create new newsfetcher for url: https://www.ladepeche.fr/rss.xml - Interval: 300000
[29.11.2021 13:35.37.438] [INFO]  Checking git for module: default
[29.11.2021 13:35.37.481] [INFO]  Checking git for module: MMM-Traffic
[29.11.2021 13:35.37.504] [INFO]  Checking git for module: MMM-Jeedom
[29.11.2021 13:35.38.791] [INFO]  Calendar-Fetcher: Broadcasting 10 events.
[29.11.2021 13:35.38.845] [INFO]  Newsfeed-Fetcher: Broadcasting 100 items.
[29.11.2021 13:35.43.285] [INFO]  Calendar-Fetcher: Broadcasting 10 events.
[29.11.2021 13:35.44.412] [ERROR] (node:3953) UnhandledPromiseRejectionWarning: Error: Command failed: cd /home/pi/MagicMirror/ && git fetch --dry-run
error: droits insuffisants pour ajouter un objet à la base de données .git/objects du dépôt
fatal: failed to write object
fatal: échec de unpack-objects

    at ChildProcess.exithandler (child_process.js:317:12)
    at ChildProcess.emit (events.js:315:20)
    at maybeClose (internal/child_process.js:1048:16)
    at Socket.<anonymous> (internal/child_process.js:439:11)
    at Socket.emit (events.js:315:20)
    at Pipe.<anonymous> (net.js:673:12)
(Use `electron --trace-warnings ...` to show where the warning was created)
[29.11.2021 13:35.44.416] [ERROR] (node:3953) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
[29.11.2021 13:35.44.418] [ERROR] (node:3953) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
^C/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
```

[... 146 lines cut ...]
				broadcastNewsUpdates: true
			}
		},
		{
			module: 'MMM-Jeedom',
			header: 'Informations Maison',
			position: "bottom_right",
			config: {
				updateInterval: 3000,
			      	jeedomAPIKey: "XXX", 
				jeedomURL: "XXX",
				jeedomPORT: XXX,
				jeedomHTTPS: XXX,
				jeedomAPIPath: "/core/api/jeeApi.php",
				sensors: [ XXX (about 15 sensors) ]
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```

**Additional Notes**: disabled the module 'updatenotification' makes the errors disappear

rejas, 2021-12-05:
Mhh... maybe the MMM-Jeedom or the MMM-Trafik module dont have a git repository associated with them? Is the error still happening when you enable the updatenotification and disable those two (or one of them)?

josselynL, 2021-12-05:
Ok just found the problem while solve another.
The owner of MagicMirror folder was root, i don't know why.

I've changed this for pi with
```
cd ~
sudo chown -R pi MagicMirror
```
And no more error...

closed 2021-12-05 by josselynL

sdetweil, 2021-12-05:
@josselynL probably did sudo git clone