MagicMirrorOrg/MagicMirror #2720: Module spacing control within config.js

chrisautrey, 2021-12-02:
Good morning,

What I have is not a bug, but an area with the formatting within the presentation layer using the config files for module layout.  Sam suggested coming here and opening an issue for it to look at for comments.  

My problem started with a hobbled together module someone made to get the PWS weather from your own WeatherUnderground data back to your mirror.  It works, but the module has some problems with with closing out the data which bleeds it into the next module and messes with it's formatting and the author looks to have gone away for a quick fix.    That's not the issue to solve here as it's a problem with that module, but as I looked at it, the problem would go away if we could get some additional spacing controls added to the config.js formatting.  This has the additional benefit where people could use this to add spacing between modules for their own mirror formatting.

What I was hoping for (that is not there today) is something like the old <br> tag that added a line break/carriage return to HTML docs.  If people could use something like this in the module section of their config files,, then they could add spacing between modules for their boards for better location granularity (and it would solve my own problem as a bonus).   Thoughts?

Below is the bleed over into the next header picture just for reference that started this where and extra line break would benefit pushing modules down the columns.  Again, not looking to solve a module problem but to look at the larger config formatting.  Thanks.  
[image]

sdetweil, 2021-12-02:
I don't see the issue. can you markup the image to highlight where the problem is

sdetweil, 2021-12-02:
for another highlighting idea, you could use cash to set the background color of the two modules to see where they end and start

chrisautrey, 2021-12-02:
I marked it up and resized it.   A normal flow of modules has the next module in the right column start the line below the one before, like "Network Devices" does at the bottom in yellow.  In this case, the text in the red circle is actually the header for the next module.   I labeled it that to fake the look on my board for now, but it's actually not part of the first module which I now realize was confusing.  

If I take the header out, the "partly cloudy"' text from the next line will move up and be jammed in and justify itself where the red circle is now instead of on the line below like the red arrow shows it should be.

chrisautrey, 2021-12-02:
I'll also look at the custom CSS later for color, but I'm 99.9% it's this module since when it is doing the API call to weatherunderground and says 'loading' there, the current temp header is below it in the proper place.  It only messes up when it does the actual drawing in of the temperature.

sdetweil, 2021-12-02:
sorry  I don't understand anything you said, jumping all over.

take the module out
disabled: true
screen shot
add the module, bad url, so always loading
screen shot
fix url
screen shot. 

I'll look at it's code when I can

chrisautrey, 2021-12-02:
OK, will put some screenshots in later and try to be clearer.  Thanks

sdetweil, 2021-12-02:
well, that was fun!.. 

I put my sample module above and below, and used the header
with module, with forced draw (set loaded=true) 
i see this 

[image]

where module A overlays the space for module B, but module B moves left?? not down??
[image]

this class is the problem child, applied to table
```
.MMM-WunderGround-PWS-Observations .large1 {
    line-height: .8cm !important;
    width: 150px;
    float: right;
}
```

if u disable width, then it behaves... 

```html

<div class="region top right">
	<div class="container" style="display: block;">
		<div id="module_11_SampleModule" class="module SampleModule SampleModule">
			<header class="module-header" style="display: block;">sample mod1 Foo Bar
			</header>
			<div class="module-content">
				<div>default message if none supplied in config.js
				</div>
			</div>
		</div>
		<div id="module_12_MMM-WunderGround-PWS-Observations" class="module MMM-WunderGround-PWS-Observations MMM-WunderGround-PWS-Observations">
			<header class="module-header" style="display: block;"> some weather
			</header>
			<div class="module-content">
				<div>
					<table class="large1">  <!------ class of interest -->
						<tr>
							<td class="pop wi wi-wind undefined">
							</td>
							<td class="popr"> undefinedKmh
							</td>
						</tr>
						<tr>
							<td class="pop wi wi-strong-wind">
							</td>
							<td class="popr"> undefinedKmh
							</td>
						</tr>
						<tr>
							<td class="pop wi wi-humidity lpad">
							</td>
							<td class="popr">undefined%
							</td>
						</tr>
						<tr>
							<td class="pop wi wi-umbrella">
							</td>
							<td class="popr"> undefinedmm
							</td>
						</tr>
						<tr>
							<td class="pop wi wi-raindrops">
							</td>
							<td class="popr">  undefinedmmh
							</td>
						</tr>
						<tr>
							<td class="pop wi wi-barometer">
							</td>
							<td class="popr">undefined
							</td>
						</tr>
						<tr>
							<td class="pop">DP
							</td>
							<td class="popr"> undefined°
							</td>
						</tr>
						<tr>
							<td class="pop">WC
							</td>
							<td class="popr"> undefined°
							</td>
						</tr>
						<tr>
							<td class="pop">HI
							</td>
							<td class="popr"> undefined°
							</td>
						</tr>
						<tr>
							<td class="pop wi wi-thermometer">
							</td>
							<td class="popr"> undefined°
							</td>
						</tr>
					</table>
				</div>
			</div>
		</div>
		<div id="module_13_SampleModule" class="module SampleModule SampleModule">
			<header class="module-header" style="display: block;">sample mod2 Foo Bar
			</header>
			<div class="module-content">
				<div>default message if none supplied in config.js
			</div>
		</div>
		</div>
	</div>
</div>
```
[image]

how did the content get outside the div tree? (without a position:absolute)

here is the config.js 
```javascript
    {
      module:"SampleModule",
      position: "top_right",
      header: "sample mod1",
      disabled: false,
      config: { 
        useHeader:true
      }
    },
{
	module: "MMM-WunderGround-PWS-Observations",
	header: " some weather",
	position: "top_right",
	config: {
    apikey:"fribble",
    pws:"01234456567",
 roundTmpDecs:1,
    sysstat: 0,
    debug: 1,
    currentweather: 1,
    wind: 1,
    humidity: 1,
    UV: 0,
    rain: 1,
    rainRate: 1,
    pressure: 1,
    dewPoint: 1,
    windChill: 1,
    heatIndex: 1,
    temperature: 1,
	}
},
    {
      module:"SampleModule",
      position: "top_right",
      header: "sample mod2",
      disabled: false,
      config: { 
	useHeader:true
      }
    }
```

shows the same in latest chrome and firefox

chrisautrey, 2021-12-02:
I have added all the screenshots as requested below showing the various conditions.

1. Problem module disabled.  All items working as expected.
[image]
2. Problem module now enabled.  This is a shot of while the MM is starting up.  This problem module is in the middle of it's API call at this point and the module below it still aligned properly.  The blacked out bar is just covering my location info.  
[image]
 3. Magic Mirror now fully loaded.  Alignment of the follow on module is now impacted.
[image]

All signs point to the problem being in the https://github.com/saabman/MMM-WunderGround-PWS-Observations module.    The author meant well, and they did get the API calls working correctly, but it's a quick and dirty solution that I am frankly just not smart enough to fix, hence looking for other options like adding lines between modules.

chrisautrey, 2021-12-02:
You were a step ahead of me as I was typing.  I will take a look at what you have here. Thanks.

sdetweil, 2021-12-02:
and if u turn off width in the class (as well as in large1)
```css
.region table {
  width: 100%;
  border-spacing: 0;
  border-collapse: separate;
}
```

it goes back to narrow again

chrisautrey, 2021-12-02:
Disabling the large section did indeed fix mine as well.  Thanks again.  I know that troubleshooting other's modules is not really what this area is for.

closed 2021-12-02 by chrisautrey