MagicMirrorOrg/MagicMirror #2725: config.js issue with end of config

blackpatina, 2021-12-06:
Hello,

Had MagicMirror working fine but wanted to edit the config.js file all looks fine but when I run the npm run config:check command it has issues with the final line..

[06.12.2021 16:25.23.364] [INFO]  Checking file...  /home/pi/MagicMirror/config/config.js
[06.12.2021 16:25.23.775] [ERROR] Your configuration file contains syntax errors :(
[06.12.2021 16:25.23.785] [ERROR] Line 106 column 2: Parsing error: Unexpected token ;

My config file with api etc removed..

What's gone wrong? I've checked it against the config.js.sample file and other config.js files but can't get it to work.

Thanks

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
        address: "localhost", // Address to listen on, can be:
                              // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
                              // - another specific IPv4/6 to listen on a specific interface
                              // - "0.0.0.0", "::" to listen on any interface
                              // Default, when address config is left out or empty, is "localhost"
        port: 8080,
        ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
                                                               // or add a specific IPv4 of 192.168.1.5 :
                                                               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:1$
                                                               // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use $
                                                               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:1$

        useHttps: false,                // Support HTTPS or not, default "false" will use HTTP
        httpsPrivateKey: "",    // HTTPS private key path, only require when useHttps is true
        httpsCertificate: "",   // HTTPS Certificate path, only require when useHttps is true

        language: "en",
        timeFormat: 24,

        units: "metric",
        // serverOnly:  true/false/"local" ,
                             // local for armv6l processors, default
                             //   starts serveronly and then starts chrome browser
                             // false, default for all  NON-armv6l devices
                             // true, force serveronly mode, because you want to.. no UI on this device

        modules: [
                {
                        module: "alert",
                },
                {
                        module: "updatenotification",
                        position: "top_bar"
                },
                {
                        module: "clock",
                        position: "top_left"
                },
                {
                        module: "calendar",
                        header: "Calendar",
                        position: "top_left",
                        config: {
                                calendars: [
                                        {
                                                symbol: "My calendar",
                                                url: "$"
                                                },
                                                {
                        module: "compliments",
                        position: "lower_third"
                },
                {
                        module: "MMM-Ring",
                        position: "middle_center",
                        config: {
                                ring2faRefreshToken: "$"
                        }
                },
                {
                        module: "currentweather",
                        position: "top_right",
                        config: {
                                location: "???",
                                locationID: "???", //ID from http://bulk.openweathermap.org/sample/city.list.jso$
                                appid: "???"
                        }
                },
                {
                        module: "weatherforecast",
                        position: "top_right",
                        header: "Weather Forecast",
                        config: {
                                location: "???",
                                locationID: "???", //ID from http://bulk.openweathermap.org/sample/city.list.jso$
                                appid: "???"
                        }
                },
                {
                        module: "newsfeed",
                        position: "bottom_bar",
                        config: {
                                feeds: [
                                        {
                                                title: "New York Times",
                                                url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
                                        }
                                ],
                                showSourceTitle: true,
                                showPublishDate: true,
                                broadcastNewsFeeds: true,
                                broadcastNewsUpdates: true
                        }
                }
        ],
};
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

sdetweil, 2021-12-07:
```
calendars: [     // <---  where is the end of the list of calendars  ]
                                    {
                                            symbol: "My calendar",
                                            url: "$"
                                            },
                                            {  // <- this is start of compliments
```

blackpatina, 2021-12-07:
Brilliant, that was my issue.. Thanks for your time

closed 2021-12-07 by blackpatina