MagicMirrorOrg/MagicMirror #2726: on forums, is there a way to blank user profile when banned?

sdetweil, 2021-12-08:
users are using forum to advertise, jamming stuff in their profile

I can delete posts and marked them banned but cannot delete the profile advertising info.
this pollutes the forum site user registry

MichMich, 2021-12-08:
What do you recommend?

sdetweil, 2021-12-08:
well, it would be nice to clear their profile.. remove the advertising that is on the profile.  we can't delete the userid, as they would just reregister it..  getting regular spam.. 

but I know nothing about the forum tool, so I don't know whats possible..

MichMich, 2021-12-08:
I have to look into it. But im flooded with work. Will get back to it. (And if not, feel free to poke me! :))

sdetweil, 2021-12-08:
I didn't mean this instant.. thats why I posted an issue.. when u(or someone else) gets around to it..

khassel, 2021-12-08:
using [this](https://forum.magicmirror.builders/user/poojaa) as example we would have to blank several fields to get rid of all possible link locations.

Found with google:
- https://community.nodebb.org/topic/14292/disable-about-me sadly with no answer
- hacking in mogodb, here example for clearing signature https://community.nodebb.org/topic/9141/deleting-all-user-signatures

So nothing out of the box so far.

sdetweil, 2022-04-17:
new reputation limits have eliminated advertising..

closed 2022-04-17 by sdetweil

MichMich, 2022-04-17:
Great to hear! Thanks!