MagicMirrorOrg/MagicMirror #2727: weatherprovider.js

winstonma, 2021-12-09:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the corresponding repository:

- karsten13/magicmirror: [https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)
- (deprecated) bastilimbach/docker-magicmirror: [https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 12 or later (recommended is 14).

**MagicMirror Version**: Please let us know which version of MagicMirror you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

khassel, 2021-12-09:
What do you mean with `make a POST API call and send some data`? Are you writing a new provider?

I looks like the `data = null` variable is not used in the current implementation.

Your code would break the tests because of `return getData(data);`.

Can you test if this code would work in your use case:

```js
	// A convenience function to make requests. It returns a promise.
	fetchData: function (url, method = "GET", data = null) {
		const getData = function (mockData) {
			return new Promise(function (resolve, reject) {
				if (mockData) {
					let mdata = mockData;
					mdata = mdata.substring(1, mdata.length - 1);
					resolve(JSON.parse(mdata));
				} else {
					const request = new XMLHttpRequest();
					request.open(method, url, true);
					request.onreadystatechange = function () {
						if (this.readyState === 4) {
							if (this.status === 200) {
								resolve(JSON.parse(this.response));
							} else {
								reject(request);
							}
						}
					};
					if (data) {
						request.setRequestHeader("Content-Type", "application/json;charset=UTF-8");
						request.send(JSON.stringify(data));
					} else {
						request.send();
					}
				}
			});
		};

		return getData(this.config.mockData);
	}
```

winstonma, 2021-12-09:
@khassel I am writing a new module and uses weather module as a reference. Yes I modified this because the `data` is not being used.

To test the function using `this.config.mockData`, should it call `fetchData(url, method = "POST", data = this.config.mockData)` instead?

khassel, 2021-12-09:
The test case doesn't use an extra provider so I used the workaround with the `getData` function. If the `config.js` has a section `mockData` the above `fetchData` function use this instead of making a fetch to an url. You can see what I implemented for the tests if you compare this file between `develop` and `master`.

So did you test if `return getData(this.config.mockData);` also works in your case?

I think `fetchData(url, method = "POST", data = this.config.mockData)` is no solution because then in `data` are different formats and we can't decide if the content is for mocking or real data.

winstonma, 2021-12-10:
Yes I tried your code and it works. I may not understand `this.config.mockData` in your use case. It just seems to me that the function `getData` looks a bit weird.

Also, I also wonder if [adding retries](https://stackoverflow.com/questions/38520438/how-to-retry-an-xhr-request-which-returns-a-promise-recursively-for-atleast-n-ti) would be great.

closed 2022-04-27 by stale[bot]