MagicMirrorOrg/MagicMirror #2728: node album creation

gripster, 2021-12-11:
hI, sorry if this has been asked have been trying to get GooglePhotos up and running, for 3 days i tried and am close to surrender. 
i cannot seem to install, node create_uploadable_album.js MyMagicMirrorAlbum.
my error is, pi@raspberrypi:~/MagicMirror $ node create_uploadable_album.js MyMagicMirrorAlbum
internal/modules/cjs/loader.js:905
  throw err;
  ^

Error: Cannot find module '/home/pi/MagicMirror/create_uploadable_album.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12)
    at internal/main/run_main_module.js:17:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}
Im sorry if this is presented wrongly, regards.

sdetweil, 2021-12-11:
sounds like you didn't do the npm install in the module folder

gripster, 2021-12-11:
hi, thanks for reply, do you know of a way to sort the issue, TIA

sdetweil, 2021-12-11:
@gripster   
> pi@raspberrypi:~/MagicMirror $ node create_uploadable_album.js MyMagicMirrorAlbum
u are not in the module folder 
```
cd ~/MagicMirror/modules/MMM-GooglePhotos 
```
then you command will work

gripster, 2021-12-11:
I will try, thanks again.

gripster, 2021-12-11:
whew! finally after 3 days, success. I just kept going over and over the install and I think I wore it down, it works but I don't know what I actually did, only something fell into place. Thanks for help, regards.

sdetweil, 2022-01-12:
@gripster   can u close this if no longer an issue

closed 2022-04-28 by stale[bot]