MagicMirrorOrg/MagicMirror #2732: snowboy issue

Hamzacos, 2021-12-17:
WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'node-pre-gyp'
Require stack:
- /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[17.12.2021 19:56.06.102] [LOG]   Loading module helpers ...
[17.12.2021 19:56.06.105] [LOG]   No helper found for module: alert.
[17.12.2021 19:56.06.108] [LOG]   Initializing new module helper ...
[17.12.2021 19:56.06.110] [LOG]   Module helper loaded: updatenotification
[17.12.2021 19:56.06.113] [LOG]   No helper found for module: clock.
[17.12.2021 19:56.06.115] [LOG]   Initializing new module helper ...
[17.12.2021 19:56.06.117] [LOG]   Module helper loaded: calendar
[17.12.2021 19:56.06.119] [LOG]   No helper found for module: compliments.
[17.12.2021 19:56.06.122] [LOG]   No helper found for module: weather.
[17.12.2021 19:56.06.125] [LOG]   Initializing new module helper ...
[17.12.2021 19:56.06.128] [LOG]   Module helper loaded: MMM-NowPlayingOnSpotify
[17.12.2021 19:56.06.130] [LOG]   Initializing new module helper ...
[17.12.2021 19:56.06.133] [LOG]   Module helper loaded: MMM-PIR-Sensor
[17.12.2021 19:56.06.136] [LOG]   Initializing new module helper ...
[17.12.2021 19:56.06.140] [LOG]   Module helper loaded: MMM-Remote-Control
[17.12.2021 19:56.06.162] [ERROR] App threw an error during load
[17.12.2021 19:56.06.171] [ERROR] Error: Cannot find module 'node-pre-gyp'
Require stack:
- /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:887:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:257:1128)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:732:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js:6:16)
    at Module._compile (internal/modules/cjs/loader.js:1078:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10)
[17.12.2021 19:56.06.178] [ERROR] Whoops! There was an uncaught exception...
[17.12.2021 19:56.06.207] [ERROR] Error: Cannot find module 'node-pre-gyp'
Require stack:
- /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:887:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:257:1128)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:732:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js:6:16)
    at Module._compile (internal/modules/cjs/loader.js:1078:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js',
    '/home/pi/MagicMirror/modules/MMM-awesome-alexa/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[17.12.2021 19:56.06.217] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[17.12.2021 19:56.06.219] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[17.12.2021 19:56.07.122] [LOG]   Launching application.
[17.12.2021 22:05.05.011] [LOG]   Shutting down server...
[17.12.2021 22:05.05.021] [LOG]   Stopping module helper: updatenotification
[17.12.2021 22:05.05.030] [LOG]   Stopping module helper: calendar
[17.12.2021 22:05.05.034] [LOG]   Stopping module helper: MMM-NowPlayingOnSpotify
[17.12.2021 22:05.05.036] [LOG]   Stopping module helper: MMM-PIR-Sensor
[17.12.2021 22:05.05.042] [ERROR] Whoops! There was an uncaught exception...
[17.12.2021 22:05.05.055] [ERROR] TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at Class.stop (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_helper.js:71:20)
    at App.stop (/home/pi/MagicMirror/js/app.js:254:16)
    at App.<anonymous> (/home/pi/MagicMirror/js/electron.js:140:7)
    at App.emit (events.js:315:20)
    at appMethod (electron/js2c/browser_init.js:69:1830)
    at Object.execute (electron/js2c/browser_init.js:69:5579)
    at MenuItem.click (electron/js2c/browser_init.js:73:1696)
    at Object.a._executeCommand (electron/js2c/browser_init.js:81:2492)
[17.12.2021 22:05.05.059] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[17.12.2021 22:05.05.063] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

sdetweil, 2021-12-18:
soundsike u didn't complete the I stall steps listed on the module wiki

sdetweil, 2021-12-27:
as I mentioned in the forum, using the script I created from the steps on the module wiki and adding the install of the libcblas dependency, the install worked