MagicMirrorOrg/MagicMirror #2733: Awesom alexa issue

Hamzacos, 2021-12-18:
WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'node-pre-gyp' Require stack: - /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_modules/snowboy/lib/node/index.js - /home/pi/MagicMirror/modules/MMM-awesome-alexa/node_helper.js - /home/pi/MagicMirror/js/app.js - /home/pi/MagicMirror/js/electron.js - /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js

sdetweil, 2021-12-18:
https://awesome-alexa.js.org/#/installation?id=step-2-install-module

sdetweil, 2021-12-18:
I made those into a script a year or so ago, hate copy/paste so much

https://www.dropbox.com/s/omrx3vztjp738hf/do-install.sh?dl=0

Hamzacos, 2021-12-18:
i follow this instruction but i have the same errure

Hamzacos, 2021-12-19:
yess i have same result

sdetweil, 2021-12-19:
sorry. no idea.
 that module hasn't been updated in years, and the author doesn't respond to issues. 

it's not a MagicMirror problem tho.

sdetweil, 2021-12-27:
i posted in #2732

closed 2022-04-27 by stale[bot]