MagicMirrorOrg/MagicMirror #2735: Several modules do not run

DigiEdAlmere, 2021-12-20:
Hi!
I created MM last week and running several modules like Weather, Calendar, NewsFeed, etc. But somehow some new modules won't start, like MMM-NASA, MMM-SolarEdge and the latest I tried was MMM-SystemStats. They all crash with this same error:

[20.12.2021 20:49.40.662] [ERROR] Whoops! There was an uncaught exception...
[20.12.2021 20:49.40.668] [ERROR] Error: Cannot find module 'request'
Require stack:
- /home/pi/MagicMirror/modules/MMM-SystemStats/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:887:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:257:1128)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:732:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-SystemStats/node_helper.js:13:15)
    at Module._compile (internal/modules/cjs/loader.js:1078:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-SystemStats/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[20.12.2021 20:49.40.671] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[20.12.2021 20:49.40.671] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[20.12.2021 20:49.40.909] [LOG]   Launching application.
[20.12.2021 20:49.53.928] [LOG]   Shutting down server...
[20.12.2021 20:49.53.934] [LOG]   Stopping module helper: updatenotification
[20.12.2021 20:49.53.936] [LOG]   Stopping module helper: calendar
[20.12.2021 20:49.53.938] [LOG]   Stopping module helper: MMM-Formula1
[20.12.2021 20:49.53.939] [LOG]   Stopping module helper: updatenotification
[20.12.2021 20:49.53.941] [LOG]   Stopping module helper: calendar
[20.12.2021 20:49.53.943] [LOG]   Stopping module helper: MMM-Formula1
[20.12.2021 20:49.53.945] [ERROR] Whoops! There was an uncaught exception...
[20.12.2021 20:49.53.949] [ERROR] TypeError: Cannot read property 'close' of undefined
    at App.stop (/home/pi/MagicMirror/js/app.js:257:14)
    at App.<anonymous> (/home/pi/MagicMirror/js/electron.js:139:7)
    at App.emit (events.js:315:20)
    at appMethod (electron/js2c/browser_init.js:69:1830)
    at Object.execute (electron/js2c/browser_init.js:69:5579)
    at MenuItem.click (electron/js2c/browser_init.js:73:1696)
    at Object.a._executeCommand (electron/js2c/browser_init.js:81:2492)

How do I fix this?

MichMich, 2021-12-20:
Try `npm install request` from within your MagicMirror folder.

DigiEdAlmere, 2021-12-20:
Thank you very much. That did it.

closed 2021-12-20 by DigiEdAlmere