MagicMirrorOrg/MagicMirror #2742: [Feature Request] Sync the Newsfeed updateInterval between modules

winstonma, 2021-12-22:
The `updateInterval` in [newsfeed](https://github.com/MichMich/MagicMirror/blob/master/modules/default/newsfeed/newsfeed.js) works perfectly. However the problem arise when there are multiple modules running on the screen and the e-ink screen is being used.

Current implementation:
```
	scheduleUpdateInterval: function () {
		this.updateDom(this.config.animationSpeed);

		// Broadcast NewsFeed if needed
		if (this.config.broadcastNewsFeeds) {
			this.sendNotification("NEWS_FEED", { items: this.newsItems });
		}

		this.timer = setInterval(() => {
			this.activeItem++;
			this.updateDom(this.config.animationSpeed);

			// Broadcast NewsFeed if needed
			if (this.config.broadcastNewsFeeds) {
				this.sendNotification("NEWS_FEED", { items: this.newsItems });
			}
		}, this.config.updateInterval);
	},
```

I tried this:
```
	scheduleUpdateInterval: function () {
		const currentTime = moment();
		const nextLoad = this.config.updateInterval - (currentTime.unix() % this.config.updateInterval);
		this.updateDom(this.config.animationSpeed);

		// Broadcast NewsFeed if needed
		if (this.config.broadcastNewsFeeds) {
			this.sendNotification("NEWS_FEED", { items: this.newsItems });
		}

		setTimeout(() => {
			this.activeItem++;
			this.scheduleUpdateInterval();
		}, nextLoad);
	},
```

This would ensure that the modules are updating in the same time, even the modules are configured separately.

rejas, 2021-12-27:
This sounds reasonable, would you like to create a PR for this feature since you already have some code? Or do you need help with that?

sdetweil, 2021-12-28:
but then u overload the network all at once with concurrent fetches...  which may impact other modules as well. 
i think the objective is a bad idea.,   

I think setTimeout should be used instead of setInterval, regardless(because if u run off the end of the interval time while processing, oops, another cycles starts, etc,etc,etc..) , but you will still end up with different cycles.  because the content is likely different between the two instances.

rejas, 2021-12-28:
> but then u overload the network all at once with concurrent fetches... which may impact other modules as well.

how about a config option for this behaviour then? something like "syncFeedUpdates" with the default being false (would be the current behaviour)? and in the docs the possible network problems could be mentioned?

winstonma, 2021-12-29:
@sdetweil `scheduleUpdateInterval` is responsible for the visual update only. The news fetching is done within [newsfeedfetcher.js](https://github.com/MichMich/MagicMirror/blob/master/modules/default/newsfeed/newsfeedfetcher.js). It is scheduled independently in the backend. Therefore the changes would not affect the new fetching portion.

winstonma, 2021-12-29:
@sdetweil I would like to add more comment on this one. The default setting of this module would fetch the news every 5 minutes, set by `reloadInterval`, and display a new news header every 10 seconds, set by `updateInterval`.

**Setup:** I set 2 of the same `newsfeed` modules on my MagicMirror, with different feeds. Also I am using e-ink screen.

**Problem:** Since the 2 instance are not in sync, therefore the visual update can be done randomly (still, they are at 10 seconds interval). It is not a problem for the traditional monitor. However e-ink screen would have a very slow refresh behavior. Therefore it would be beneficial to group all visual updates in packs and make sure all refresh are done in one go.

MichMich, 2021-12-29:
I understand the need for this for epaper displays, but for regular mirrors this takes away a lot of the visual dynamics. I think a better approach would be to make a change to the core to set a minimum time between updates. Since the core is in charge of the getDom calls, it should be possible to make the core in charge of this. This way we can add a config option like `minimumTimeBetweenUIUpdates`. I don't think we need to clutter the modules with this.

winstonma, 2021-12-30:
@MichMich I tried this on regular mirrors and I don't see there are any changes on the visual dynamics. I would like to describe a bit more.

For example, when the `updateInterval` is set to 10 seconds, the original implementation would display a new header every 10 seconds (start at anytime). But my modification would ensure that the header would be displayed in every 10th second (xx:10, xx:20, xx:30). So that if multiple modules is implemented and the modules have the same `updateInterval`, then they would get updated at the same time.

I guess the visual dynamics is controlled by `animationSpeed` which is not affected.

closed 2022-04-28 by stale[bot]