MagicMirrorOrg/MagicMirror #2743: format config as YAML

CFenner, 2021-12-22:
What do you think is the effort to support a YAML based config file in MM? I often have typos in the JSON and think that YAML would be a lot easier to maintain.

MichMich, 2021-12-22:
Nah, not worth it.

This is one of the reasons: https://docs.magicmirror.builders/getting-started/manifesto.html

MichMich, 2021-12-22:
And with that I mean, taking away hurdles like json doesn't teach you anything.

rejas, 2021-12-22:
Doesnt yaml have the problem of requiring correct intendation?

sdetweil, 2021-12-22:
every language has its quirks. for the untrained they are all mine fields

thats why I created the form based config.  config should not be like this. its the least important part of the solution, and currently causes the most trouble

khassel, 2021-12-22:
> I often have typos in the JSON

Its no JSON but a javascript object.

> and think that YAML would be a lot easier to maintain.

That depends on the person using it.

> Doesnt yaml have the problem of requiring correct intendation?

corrrect, and if you ever had a `tab` instead of spaces in a `yaml` file you often get no leading error message from the tools using this incorrect file ... real fun.

> every language has its quirks. for the untrained they are all mine fields

👍