MagicMirrorOrg/MagicMirror #2749: is there any doc on useHttps?  I can't find any

sdetweil, 2021-12-25:
user trying to set it up... but not referenced in config docs

and the config sample 

path to  implies, NOT filename

sdetweil, 2021-12-25:
this post  says the code is broken 
https://forum.magicmirror.builders/topic/14091/issues-enabling-https-on-rp-4?_=1640398160931
```
useHttps: false -> useHttps: true

Nevertheless I only got a black screen afterwards.

I tried everything I could think of, but, being that Im not much of a programmer, it basically meant hitting the keyboard with one hand while having another coffee hoping that would do it.

Eventually, I managed to find out that I could connect to the https://localhost:port via a browser, which made me realise the keyboard-hitting works… and that it was running… just not with electron.

Well, tonight, after many an hour in front of this, I was able to get it to work, so I can narrow it down to the following:

Create your certificates correctly
Make sure you put the right address where they are located
Not enough coffee
There is a piece of code in electron.js that seems to be checking the wrong variable:
// if (config[“tls”] !== null && config[“tls”]) {
if (config.useHttps) {
prefix = “https://”;
} else {
prefix = “http://”;
}

The line in “//” was commented out by me. I couldnt find much of a reference to tls anywhere.

Finally, as the certificate is self signed, these lines had to be added also to electron.js:

app.on(‘certificate-error’, (event, webContents, url, error, certificate,
callback) => {
event.preventDefault();
callback(true);
});

```

I see electron.js in develop is still checking config['tls'], not config.useHttps

khassel, 2021-12-25:
yes, this is a mess.

AFAIK there were 2 use cases implemented using `https` so far:

- option `useHttps` for `serveronly` mode: parameter `useHttps` and certs in `config.js`
- option `tls` for `clientonly` mode: runtime parameter `--use-tls` needed

But I never understood any of the https use cases. Why encrypt things when the whole config per default is public and so readable by erveryone (including config secrets)? You only have to call `http(s)://<ip-of-mm>:<port-of-mm>/config/`.

So if anyone wants to have a public MagicMirror-Url he should protect this url with some sort of authentification and therefore simplest solution is a reverse proxy which also can handle the https part out of the box.

sdetweil, 2021-12-25:
apparently home assistant in https mode cannot access mm in http mode

khassel, 2021-12-25:
to get this running without breaking the existing code we could change the `if` statement in `electron.js` to

```js
	if ((config["tls"] !== null && config["tls"]) || config.useHttps) {
		prefix = "https://";
	} else {
		prefix = "http://";
	}
```

and add the `app.on(‘certificate-error’ ...)`

sdetweil, 2021-12-26:
@khassel   thanks.. I have sent the user an updated electron.js with those changes

khassel, 2021-12-26:
someone should decide if this should be done in the code here, can provide a PR but only if this is wanted.

So any comments to this @sdetweil @rejas @MichMich ?

sdetweil, 2021-12-26:
I think we want it. I have code ready for the pr, have sent to user for testing. it's the holidays, so not sure on his responsiveness

rejas, 2021-12-27:
> So any comments to this @sdetweil @rejas @MichMich ?

Got no experience with the https stuff in general so I follow @sdetweil's lead here :-)

PR #2756 (2021-12-27): enable useHttps for full electron version

> - Does the pull request solve a **related** issue?
> - If so, can you reference the issue like this `Fixes #<issue_number>`?
Fixes: #2749 
updated  CHANGELOG.md file.