MagicMirrorOrg/MagicMirror #2752: calendar full day event one day to early

khassel, 2021-12-25:
ics file:

```
BEGIN:VCALENDAR
PRODID:-//Google Inc//Google Calendar 70.9054//EN
VERSION:2.0
CALSCALE:GREGORIAN
METHOD:PUBLISH
X-WR-CALNAME:Geburtstage
X-WR-TIMEZONE:Europe/Berlin
X-WR-CALDESC:
BEGIN:VTIMEZONE
TZID:Europe/Berlin
X-LIC-LOCATION:Europe/Berlin
BEGIN:DAYLIGHT
TZOFFSETFROM:+0100
TZOFFSETTO:+0200
TZNAME:CEST
DTSTART:19700329T020000
RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=-1SU
END:DAYLIGHT
BEGIN:STANDARD
TZOFFSETFROM:+0200
TZOFFSETTO:+0100
TZNAME:CET
DTSTART:19701025T030000
RRULE:FREQ=YEARLY;BYMONTH=10;BYDAY=-1SU
END:STANDARD
END:VTIMEZONE
BEGIN:VEVENT
DTSTART;VALUE=DATE:20160205
DTEND;VALUE=DATE:20160206
RRULE:FREQ=YEARLY
DTSTAMP:20211225T201951Z
UID:bc4daa6b-a578-4d99-8d2e-7f2f4349455e
CREATED:20170127T230413Z
DESCRIPTION:
LAST-MODIFIED:20170224T234614Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Birthday-Test
TRANSP:TRANSPARENT
X-MOZ-GENERATION:1
END:VEVENT
END:VCALENDAR
```

config.js:
```js
    {
      module: "calendar",
      header: "Termine",
      position: "top_left",
      config: {
        maximumEntries: 18,
        maximumNumberOfDays: 70,
        colored: true,
        calendars: [
          {
             symbol: "hand-o-right",
            url: "http://localhost:8080/config/basic.ics"
          },
        ]
      }
    },
```

I would expect to get Feb. 5th. but the calendar displays Feb. 4th.

Here the debug logs:

```bash
[25.12.2021 22:37.05.808] [LOG]   Create new calendarfetcher for url: http://localhost:8080/config/basic.ics - Interval: 300000
[25.12.2021 22:37.05.818] [LOG]   Create new newsfetcher for url: https://www.polizei.sachsen.de/vwd/LMSSN-Bestand.RSS_STD.xml - Interval: 300000
[25.12.2021 22:37.05.886] [INFO]  Checking git for module: default
[25.12.2021 22:37.05.926] [DEBUG] parsed data={"5420a098-a968-44a1-b673-c39897f4aaea":{"type":"VTIMEZONE","params":[],"tzid":"Europe/Berlin","LIC-LOCATION":"Europe/Berlin","f9a1fa3c-6c40-4c96-89b8-366fe5103ffa":{"type":"DAYLIGHT","params":[],"tzoffsetfrom":"+0100","tzoffsetto":"+0200","tzname":"CEST","start":"1970-03-29T01:00:00.000Z","datetype":"date-time","rrule":"RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=-1SU","end":"1970-03-29T01:00:00.000Z"},"4d158d20-2715-4669-9002-06d4ccadbdef":{"type":"STANDARD","params":[],"tzoffsetfrom":"+0200","tzoffsetto":"+0100","tzname":"CET","start":"1970-10-25T02:00:00.000Z","datetype":"date-time","rrule":"RRULE:FREQ=YEARLY;BYMONTH=10;BYDAY=-1SU","end":"1970-10-25T02:00:00.000Z"},"end":"2021-12-26T21:37:05.921Z"},"bc4daa6b-a578-4d99-8d2e-7f2f4349455e":{"type":"VEVENT","params":[],"start":"2016-02-05T00:00:00.000Z","datetype":"date","end":"2016-02-05T23:00:00.000Z","rrule":{"_cache":{"all":false,"before":[],"after":[],"between":[]},"origOptions":{"dtstart":"2016-02-05T00:00:00.000Z","freq":0},"options":{"freq":0,"dtstart":"2016-02-05T00:00:00.000Z","interval":1,"wkst":0,"count":null,"until":null,"bysetpos":null,"bymonth":[2],"bymonthday":[5],"bynmonthday":[],"byyearday":null,"byweekno":null,"byweekday":null,"bynweekday":null,"byhour":[0],"byminute":[0],"bysecond":[0],"byeaster":null}},"dtstamp":"2021-12-25T20:19:51.000Z","uid":"bc4daa6b-a578-4d99-8d2e-7f2f4349455e","created":"2017-01-27T23:04:13.000Z","description":"","lastmodified":"2017-02-24T23:46:14.000Z","location":"","sequence":"0","status":"CONFIRMED","summary":"Birthday-Test","transparency":"TRANSPARENT","MOZ-GENERATION":"1"}}
[25.12.2021 22:37.05.926] [DEBUG] There are 2 calendar entries.
[25.12.2021 22:37.05.927] [DEBUG] Processing entry...
[25.12.2021 22:37.05.928] [DEBUG] Processing entry...
[25.12.2021 22:37.05.928] [DEBUG] Event:
{"type":"VEVENT","params":[],"start":"2016-02-05T00:00:00.000Z","datetype":"date","end":"2016-02-05T23:00:00.000Z","rrule":{"_cache":{"all":false,"before":[],"after":[],"between":[]},"origOptions":{"dtstart":"2016-02-05T00:00:00.000Z","freq":0},"options":{"freq":0,"dtstart":"2016-02-05T00:00:00.000Z","interval":1,"wkst":0,"count":null,"until":null,"bysetpos":null,"bymonth":[2],"bymonthday":[5],"bynmonthday":[],"byyearday":null,"byweekno":null,"byweekday":null,"bynweekday":null,"byhour":[0],"byminute":[0],"bysecond":[0],"byeaster":null}},"dtstamp":"2021-12-25T20:19:51.000Z","uid":"bc4daa6b-a578-4d99-8d2e-7f2f4349455e","created":"2017-01-27T23:04:13.000Z","description":"","lastmodified":"2017-02-24T23:46:14.000Z","location":"","sequence":"0","status":"CONFIRMED","summary":"Birthday-Test","transparency":"TRANSPARENT","MOZ-GENERATION":"1"}
[25.12.2021 22:37.05.928] [DEBUG] start: Fri Feb 05 2016 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.928] [DEBUG] end:: Sat Feb 06 2016 00:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.929] [DEBUG] duration: 82800000
[25.12.2021 22:37.05.929] [DEBUG] title: Birthday-Test
[25.12.2021 22:37.05.929] [DEBUG] fullday
[25.12.2021 22:37.05.929] [DEBUG] pastLocal: Sat Dec 25 2021 00:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.930] [DEBUG] futureLocal: Fri Mar 04 2022 23:59:59 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.930] [DEBUG] Search for recurring events between: Sat Dec 25 2021 00:00:00 GMT+0100 (Central European Standard Time) and Fri Mar 04 2022 23:59:59 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.937] [DEBUG] Title: Birthday-Test, with dates: ["2022-02-05T00:00:00.000Z"]
[25.12.2021 22:37.05.937] [DEBUG] event.recurrences: undefined
[25.12.2021 22:37.05.937] [DEBUG]  recurring date is Sat Feb 05 2022 01:00:00 GMT+0100 (Central European Standard Time) offset is -60
[25.12.2021 22:37.05.937] [DEBUG]  recurring date is Sat Feb 05 2022 01:00:00 GMT+0100 (Central European Standard Time) offset is -1 Hour is 01
[25.12.2021 22:37.05.938] [DEBUG] Fullday
[25.12.2021 22:37.05.938] [DEBUG] new recurring date1 is Fri Feb 04 2022 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.939] [DEBUG] Corrected startDate: Fri Feb 04 2022 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.939] [DEBUG]  if no tz, guess based on now
[25.12.2021 22:37.05.953] [DEBUG] initial tz=Europe/Berlin
[25.12.2021 22:37.05.953] [DEBUG] corrected tz=Europe/Berlin
[25.12.2021 22:37.05.954] [DEBUG] start date/time=Fri Feb 05 2016 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.956] [DEBUG] start offset=60
[25.12.2021 22:37.05.956] [DEBUG] start date/time w tz =Fri Feb 05 2016 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.957] [DEBUG] event date=Fri Feb 04 2022 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.957] [DEBUG] event offset=60 hour=1 event date=Fri Feb 04 2022 01:00:00 GMT+0100 (Central European Standard Time)
[25.12.2021 22:37.05.957] [DEBUG] adjustHours=0
[25.12.2021 22:37.05.958] [DEBUG] duration: 82800000
[25.12.2021 22:37.05.959] [DEBUG] saving event: false
[25.12.2021 22:37.05.959] [INFO]  Calendar-Fetcher: Broadcasting 1 events.
```

I found this code in `calendarfetcher.js`:

```js
						// For full day events, the time might be off from RRULE/Luxon problem
						// Get time zone offset of the rule calculated event
						let dateoffset = date.getTimezoneOffset();
```

Could be the reason for the wrong calculation but to be honest I have no clue what is done there ... May I forgot something in the config section to get this working correctly?

khassel, 2021-12-25:
@sdetweil can you please take a look at this because the `calendarutils.js` from before [your PR](https://github.com/MichMich/MagicMirror/pull/2748) is working correctly, thanks.

sdetweil, 2021-12-26:
i see it.. thinking

the problem is that luxon, the rrule processor expects the date (fullday) to be in local format, not UTC..

jupadin, 2021-12-26:
I think, as mentioned in the other PR, this is related to the calculation of the DST (which currently uses the times from 2007, respectively from 2013 instead from the current year.
See:
https://github.com/MichMich/MagicMirror/blob/222a5f37791d9f8ae6182ef03487fdd509f1954f/modules/default/calendar/calendarutils.js#L76-L77

and 

https://www.timeanddate.com/time/change/usa

sdetweil, 2021-12-26:
not really a DST thing. it's a luxon problem

jupadin, 2021-12-26:
Okay, then different problem 😕

sdetweil, 2021-12-26:
yeh, the problem is that date

using Australia for example the date/time is

1225:0000 in Sydney for Christmas morning
but that is 1224:1300 utc

so luxon uses the date, 1224 and produces a recurring date off by a day

similarly for west of utc

for few hours my Austin time is
1225:0000, which is utc 1225:0600

so both cause problems 

we don't know the date was mangled,  iCal returned utc time. 
then we ask for rule matching date/times between now and a year from now.

but if start is inside the utc offset then the result dates will be wrong. 

the problem child is the upper boundary
utc+11, which which utc date is that,? 1225 or 1224

khassel, 2021-12-26:
@sdetweil wrote a test for recurring events which iterates over timezones.

So with the old version of `calendarutils.js` all tests are green, with your last changes only UTC <=0 is working.

```bash
    Recurring event per timezone
      ✕ should contain text "Mar 25th" in timezone UTC 12 (8 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 11 (2 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 10 (1 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 9 (1 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 8 (5 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 7 (3 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 6 (1 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 5 (2 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 4 (1 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 3 (2 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 2 (3 ms)
      ✕ should contain text "Mar 25th" in timezone UTC 1 (3 ms)
      ✓ should contain text "Mar 25th" in timezone UTC 0 (3 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -1 (3 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -2 (2 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -3 (1 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -4 (1 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -5 (2 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -6 (1 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -7 (2 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -8 (2 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -9 (2 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -10 (1 ms)
      ✓ should contain text "Mar 25th" in timezone UTC -11 (2 ms)
```

PR #2755 (2021-12-27): fix date correction comparison for fulldate, just less than, not less than or equal

> - Does the pull request solve a **related** issue?
fixes: #2752
closed 2021-12-29 by khassel