MagicMirrorOrg/MagicMirror #2757: avoiding merge conflicts in changelog

khassel, 2021-12-27:
Having several PR's in parallel results often in merge conflicts because they change the same line in the `CHANGELOG.md` file.

There are solutions to solve this in the range from

- e.g. [chalog](https://github.com/jthomperoo/chalog) which would add a step before releasing to build the current `CHANGELOG.md` from md-files committed with every PR

to

- e.g. [Github Workflow](https://github.com/marketplace/actions/release-changelog-builder) which comes without a `CHANGELOG.md` file and instead reads the information out of the PR's

Before investigating in this I like to have feedback if this should be done, maybe others are happy with the current solution.

sdetweil, 2021-12-27:
this would be great..

I created sep branches for my three changes..
with 3 changelog updates..
and 3 PRs  

don't know what caused this conflict

khassel, 2021-12-27:
> don't know what caused this conflict

this is good explained [here](https://about.gitlab.com/blog/2018/07/03/solving-gitlabs-changelog-conflict-crisis/)

DanielHabenicht, 2022-01-09:
If you are fine with each changelog entry being an extra commit and enforcing semver this could also be an option: https://github.com/semantic-release/semantic-release 

I am using it for my module (and many other projects): https://github.com/DanielHabenicht/MMM-json
Happy to provide a PR if you are interested. 
The downside (or upside) is that each PR commit message would have to adhere the semver format. (But there are workflows for checking it before committing, see [here for a bad example](https://github.com/CovOpen/CovQuestions/pull/196#issuecomment-864420271))
Upside: Never think about release numbers nor manual adding tags or github releases.

khassel, 2022-01-09:
@DanielHabenicht thanks. My goal is to get rid of this merge conflicts, the above tools are proposals, have no experiences with them. So I'm open to use `semantic-release` but the general decision has to be made by @MichMich , he has to work with it ...

MichMich, 2022-01-09:
As much as I hate merge conflicts I don't think we should overcomplicate things. Partly because this is one more hurdle to newcomers. 

Also: if we use additional tools they might not be around in the long run. 

I'm all for a nice solution as long as it doesn't add addition dependencies, it doesn't require extra work for contributors and it doesn't change our current versioning scheme. 😅

DanielHabenicht, 2022-01-09:
Well, it is up to you, as I said, I am happy to provide a PR and I for sure love semantic release but in the end, it depends if it fits the community (and I can't decide that, because I am fairly new) 
It is been actively maintained for 4 years and as long as Javascript is around I think it will be around. 😏 

Configuration wise it would add: 
- [`releaserc.json`](https://github.com/DanielHabenicht/MMM-json/blob/main/.releaserc.json)
- [semantic-release dev dependencies](https://github.com/DanielHabenicht/MMM-json/blob/e27d9cd95582fb2ff7b8537b7969085118cd90a0/package.json#L26-L30)
- [release workflow](https://github.com/DanielHabenicht/MMM-json/blob/main/.github/workflows/release.yml)
- [a title check worflow](https://github.com/DanielHabenicht/MMM-json/blob/main/.github/workflows/pr-title-check.yml)
- Personally, I would add another workflow for showing the release number that will be generated, so one can be sure that the title is right. 

Regarding extra work, it would actually benefit the contributors. Instead of changing the `changelog.md` (and resolving merge conflicts) they only have to specify their changes in the title and comment section of the Pull Request (downside: you should only allow squash commits). 
The Release would be automatically generated afterwards. As you are already using semver (as semantic release does) the versioning scheme does not change.  
You would just need to change the title to something like "Twenty Twenty Two!" manually - because only humans do that 😄.

MichMich, 2022-01-09:
The thing is that we have a release cycle every 3 months. The reason for this is that a mirror is not something you want to update/maintain weekly. So I try to keep the updates to an absolute minimum. Every 3 months I release all the contributions in the past three months.

DanielHabenicht, 2022-01-09:
That's also possible in that case you can make the release workflow manual and trigger it when you want to. (Or only on the merge to the specific branch)

If I am honest I do not update my mirror at all.  Only if I add a new module and want to get a new feature. 
In that case I would have to wait for you to release it or just switch to the development branch as the whole project is just a git repo. 😅

MichMich, 2022-01-09:
I appreciate your suggestion. But I think we are trying to solve something that isn't really an issue in this repo. 

Most conflicts are not due to the changelog. If they are just the changelog I can easily fix them using the GitHub UI. 

Also, most conflicts are most of the time solved pretty quick, if not it's usually a great learning opportunity for the contributor. 

So I don't think a change is necessary.

closed 2022-01-09 by khassel