MagicMirrorOrg/MagicMirror #2759: what npm level do we need for 2.18?

sdetweil, 2021-12-31:
node going up to 16 for electron

khassel, 2021-12-31:
don't know if there is a need for a specific version, the versions are listed [here](https://nodejs.org/en/download/releases/)

sdetweil, 2021-12-31:
ok, that table was useful... 

for electron 16.0.5, its node 16 , npm 7.10 (down some) or 7.11.2 (up some)

```
Node.js 16.1.0 |   | 2021-05-04 | 9.0.257.24 | 7.11.2 | 93
Node.js 16.0.0 |   | 2021-04-20 | 9.0.257.17 | 7.10.0 | 93
```

khassel, 2021-12-31:
Are you mixing up node and electron version?

AFAIK electron v16 needs no special node version, electron supports latest 4 node versions concerning [this post](https://www.electronjs.org/blog/electron-16-0).

If you download latest node v16 you will get v16.13.1 with npm 8.1.2.

sdetweil, 2021-12-31:
electron 16 says it uses node 16. away for rest of day.

sdetweil, 2022-01-01:
from 
https://www.npmjs.com/package/electron-releases

Tag | Published | npm | Prerelease | Module Version | Node | Chrome | Downloads
-- | -- | -- | -- | -- | -- | -- | --
v16.0.5 | 2021-12-17 | latest, 16-x-y | no | 99 | 16.9.1 | 96.0.4664.55 | 66317
-- | -- | -- | -- | -- | -- | -- | --

Electron V16.0.5 is the required version in MM package.json

khassel, 2022-01-01:
I know this table but it's not clear to me what the Node Version here means, may this is the version electron was built with.

AFAIK there is no statement that this version is required, reading [this document](https://www.electronjs.org/docs/latest/tutorial/support#supported-versions) electron v16 is working with all node versions between 13 and 16.

sdetweil, 2022-01-01:
that document  says what Electron versions you can get from the npmjs library (using npm) .. rolling  current -2. 

it doesn't say anything about the supporting environment  required to RUN a specific version of Electron

which I think comes from the releases table. 
(I also don't know what the npm column means, as it is mostly empty)

bugsounet, 2022-01-01:
@sdetweil , @khassel 

don't take my comment for you :)

```
> @bugsounet/pir@1.0.9 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/pir
> MagicMirror-rebuild

✔ MagicMirror Rebuild Complete
```

on launch library:
```
[01.01.2022 23:35.56.599] [ERROR] [GA] @bugsounet/pir Loading error! Error: The module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 99. Please try re-compiling or re-installing
```

it's not 93 but 99 ;)

powaaa we have to recode a lot of things :/

Yes .. I assume to force use node v14.x ... now v16.x ... and specific version not a real solution :/

sdetweil, 2022-01-01:
yes, pir is a pain all the time.

couple other binaries around too. saw one on keybindings

bugsounet, 2022-01-01:
@sdetweil : it's a sample :)
Personnaly I rebuild some personal library for MM using :) [Impact is the same]

bugsounet, 2022-01-02:
@sdetweil: finaly recoded without problem ;)

used: [my magicmirror-rebuild npm package](https://github.com/bugsounet/MagicMirror-rebuild)

sdetweil, 2022-01-11:
using 16.1.0 for node, whatever comes along for npm

closed 2022-01-11 by sdetweil