MagicMirrorOrg/MagicMirror #2765: Pm2 Log Error - How can i solve these errors ?

Fredi009sg, 2022-01-02:
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:46.51.216] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
Tue, 28 Dec 2021 12:46:51 GMT express deprecated res.send(status): Use res.sendStatus(status) instead at modules/MMM-Remote-Control/node_helper.js:138:25
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:48.37.248] [ERROR] [31mWARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: /home/pi/MagicMirror/config/config.js:90[39m
[31m		{[39m
[31m		^[39m

[31mSyntaxError: Unexpected token '{'[39m
[31m    at wrapSafe (internal/modules/cjs/loader.js:986:16)[39m
[31m    at Module._compile (internal/modules/cjs/loader.js:1034:27)[39m
[31m    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10)[39m
[31m    at Module.load (internal/modules/cjs/loader.js:935:32)[39m
[31m    at Module._load (internal/modules/cjs/loader.js:776:14)[39m
[31m    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)[39m
[31m    at Module.require (internal/modules/cjs/loader.js:959:19)[39m
[31m    at require (internal/modules/cjs/helpers.js:88:18)[39m
[31m    at loadConfig (/home/pi/MagicMirror/js/app.js:69:14)[39m
[31m    at App.start (/home/pi/MagicMirror/js/app.js:212:3)[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:50.25.169] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:52.34.401] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:54.24.598] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:56.15.696] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 13:57.12.836] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:03.35.703] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
[28.12.2021 14:12.06.071] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:19.11.743] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:21.01.131] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:22.10.511] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:23.36.657] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:25.07.967] [ERROR] [31mWARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: /home/pi/MagicMirror/config/config.js:143[39m
[31m{[39m
[31m^[39m

[31mSyntaxError: Unexpected token '{'[39m
[31m    at wrapSafe (internal/modules/cjs/loader.js:986:16)[39m
[31m    at Module._compile (internal/modules/cjs/loader.js:1034:27)[39m
[31m    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10)[39m
[31m    at Module.load (internal/modules/cjs/loader.js:935:32)[39m
[31m    at Module._load (internal/modules/cjs/loader.js:776:14)[39m
[31m    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)[39m
[31m    at Module.require (internal/modules/cjs/loader.js:959:19)[39m
[31m    at require (internal/modules/cjs/helpers.js:88:18)[39m
[31m    at loadConfig (/home/pi/MagicMirror/js/app.js:69:14)[39m
[31m    at App.start (/home/pi/MagicMirror/js/app.js:212:3)[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:26.33.999] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:28.47.516] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:29.16.076] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[28.12.2021 14:31.13.222] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
Gdk-Message: 23:00:00.970: electron: Fatal IO error 11 (Die Ressource ist zur Zeit nicht verfügbar) on X server :0.

XIO:  fatal IO error 62 (Der virtuelle Zeitnehmer ist abgelaufen) on X server ":0"
      after 11902688 requests (11902688 known processed) with 0 events remaining.
[28.12.2021 23:02.28.676] [WARN]  [33mYou're using a full whitelist configuration to allow for all IPs[39m
[... 68 lines cut ...]

[02.01.2022 20:10.50.593] [ERROR] (node:3251) UnhandledPromiseRejectionWarning: Error: fatal: Konnte '/home/pi/MagicMirror/modules/MMM-MyGarbage/.git/index.lock' nicht erstellen: Keine Berechtigung

    at Object.action (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/plugins/error-detection.plugin.js:30:33)
    at PluginStore.exec (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/plugins/plugin-store.js:21:33)
    at /home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:95:45
    at new Promise (<anonymous>)
    at GitExecutorChain.handleTaskData (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:93:16)
    at GitExecutorChain.<anonymous> (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:77:46)
    at Generator.next (<anonymous>)
    at fulfilled (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:5:58)
(Use `electron --trace-warnings ...` to show where the warning was created)
[02.01.2022 20:10.50.596] [ERROR] (node:3251) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
[02.01.2022 20:10.50.601] [ERROR] (node:3251) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[02.01.2022 20:10.51.843] [ERROR] (node:3251) UnhandledPromiseRejectionWarning: Error: fatal: Konnte '/home/pi/MagicMirror/modules/MMM-MyGarbage/.git/index.lock' nicht erstellen: Keine Berechtigung

    at Object.action (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/plugins/error-detection.plugin.js:30:33)
    at PluginStore.exec (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/plugins/plugin-store.js:21:33)
    at /home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:95:45
    at new Promise (<anonymous>)
    at GitExecutorChain.handleTaskData (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:93:16)
[... cut ...]
    at fulfilled (/home/pi/MagicMirror/modules/MMM-Remote-Control/node_modules/simple-git/src/lib/runners/git-executor-chain.js:5:58)
[02.01.2022 20:10.51.844] [ERROR] (node:3251) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)

sdetweil, 2022-01-02:
what mm version?

grep version ~/MagicMirror/package.json

what node and npm version

node -v
npm -v

what os?
lsb_release -a

how did u get here? install today, or upgrade

in either case how did u do it? manual or automated?

Fredi009sg, 2022-01-03:
what mm version?
**2.18.0**
grep version ~/MagicMirror/package.json

what node and npm version

node -v
**v16.13.1**
npm -v
**8.1.2**
what os?
lsb_release -a

how did u get here? install today, or upgrade
**i installed it with the script from your site https://github.com/sdetweil/MagicMirror_scripts**
in either case how did u do it? manual or automated?

sdetweil, 2022-01-03:
also, could you post ~/install.log

I just redid install on bullseye and it worked fine.. same node/npm/pm2

```
~/MagicMirror $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

Fredi009sg, 2022-01-03:
> thanks.. can u give results of
> 
> lsb_release -a

No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

Fredi009sg, 2022-01-03:
> also, could you post ~/install.log
> 
> I just redid install on bullseye and it worked fine.. same node/npm/pm2
> 
> ```
> ~/MagicMirror $ lsb_release -a
> No LSB modules are available.
> Distributor ID:	Raspbian
> Description:	Raspbian GNU/Linux 11 (bullseye)
> Release:	11
> Codename:	bullseye
> ```

install starting  - Di Dez 28 12:51:23 CET 2021
installing on armv7l processor system
the os is Distributor ID: Raspbian Description: Raspbian GNU/Linux 10 (buster) Release: 10 Codename: buster
[96mInstalling helper tools ...[90m
Paketlisten werden gelesen...
Abhängigkeitsbaum wird aufgebaut....
Statusinformationen werden eingelesen....
build-essential ist schon die neueste Version (12.6).
curl ist schon die neueste Version (7.64.0-4+deb10u2).
git ist schon die neueste Version (1:2.20.1-2+deb10u3).
unzip ist schon die neueste Version (6.0-23+deb10u2).
wget ist schon die neueste Version (1.20.1-1.1).
Vorgeschlagene Pakete:
  doc-base
Die folgenden NEUEN Pakete werden installiert:
  pv
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Es müssen 46,4 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 113 kB Plattenplatz zusätzlich benutzt.
Holen:1 https://archive.raspbian.org/raspbian buster/main armhf pv armhf 1.6.6-1 [46,4 kB]
Es wurden 46,4 kB in 0 s geholt (160 kB/s).
Vormals nicht ausgewähltes Paket pv wird gewählt.
(Lese Datenbank ... 
(Lese Datenbank ... 5%
(Lese Datenbank ... 10%
(Lese Datenbank ... 15%
(Lese Datenbank ... 20%
(Lese Datenbank ... 25%
(Lese Datenbank ... 30%
(Lese Datenbank ... 35%
(Lese Datenbank ... 40%
(Lese Datenbank ... 45%
(Lese Datenbank ... 50%
(Lese Datenbank ... 55%
(Lese Datenbank ... 60%
(Lese Datenbank ... 65%
(Lese Datenbank ... 70%
(Lese Datenbank ... 75%
(Lese Datenbank ... 80%
(Lese Datenbank ... 85%
(Lese Datenbank ... 90%
(Lese Datenbank ... 95%
(Lese Datenbank ... 100%
(Lese Datenbank ... 166302 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../archives/pv_1.6.6-1_armhf.deb ...
Entpacken von pv (1.6.6-1) ...
pv (1.6.6-1) wird eingerichtet ...
Trigger für man-db (2.8.5-2) werden verarbeitet ...
[96mUpdating packages ...[90m
OK:1 http://phoscon.de/apt/deconz buster InRelease OK:2 http://archive.raspberrypi.org/debian buster InRelease OK:3 http://raspbian.raspberrypi.org/raspbian buster InRelease OK:4 https://repos.influxdata.com/debian buster InRelease OK:5 https://archive.raspbian.org/raspbian buster InRelease OK:6 https://deb.nodesource.com/node_16.x buster InRelease Paketlisten werden gelesen...
apt-get update  completed ok
apt-get upgrade  started
apt upgrade result =rc=0 Paketlisten werden gelesen...
Abhängigkeitsbaum wird aufgebaut....
Statusinformationen werden eingelesen....
Paketaktualisierung (Upgrade) wird berechnet...
Die folgenden Pakete sind zurückgehalten worden:
[... 130 lines cut ...]

Target path
/etc/systemd/system/pm2-pi.service
Command list
[ 'systemctl enable pm2-pi' ]
[PM2] Writing init configuration in /etc/systemd/system/pm2-pi.service
[PM2] Making script booting at startup...
[PM2] [-] Executing: systemctl enable pm2-pi...
[PM2] [v] Command successfully executed.
+---------------------------------------+
[PM2] Freeze a process list on reboot via:
$ pm2 save

[PM2] Remove init script via:
$ pm2 unstartup systemd
pm2 startup command done
configure the pm2 config file for MagicMirror
start MagicMirror via pm2 now
save MagicMirror pm2 config now
stop MagicMirror via pm2 now
gsettings screen saver already disabled
screensaver via lightdm already disabled
lxsession screen saver already disabled
[92mWe're ready! Run [1m[97mpm2 start MagicMirror[0m[92m from the ~/MagicMirror directory to start your MagicMirror.[0m
install completed - Di Dez 28 13:35:45 CET 2021

sdetweil, 2022-01-03:
how did u get node 16 installed?  my script (before Saturday) only installed node 14

sdetweil, 2022-01-03:
also, output of uname -a

Fredi009sg, 2022-01-03:
> how did u get node 16 installed? my script (before Saturday) only installed node 14

i had a previous installation before but i deleted it because there were the same issue with the blackscreen after a while and i want to give it a try with a fresh installation

Fredi009sg, 2022-01-03:
> 

Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

Fredi009sg, 2022-01-03:
> > 
> 
> Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

Fredi009sg, 2022-01-03:
> also, output of uname -a

Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

sdetweil, 2022-01-03:
did u accidentally close? 

the answer for how you got node 16 didn't help me..   

did u install node manually before starting my install???

Fredi009sg, 2022-01-03:
> > did u accidentally close?
> 
> yes i reopen it
> 
> > the answer for how you got node 16 didn't help me..
> 
> > did u install node manually before starting my install???
> 
> could be that i installed it with the previous installation manually

sdetweil, 2022-01-03:
hm.. i can't reproduce it.. 
reflashed buster (legacy) 
manually copied  2.17.1 (clone gets 2.18)
changed local copy of install script to not clone, as I already had it..
install script installed node v16.13.1 and npm 8.1.2 
then installed  MM and pm2, and started mm ok, with initial config.js

what pi is this?  free -m might tell me.. 
i have a 4 gig pi 4 
```
free -m
              total        used        free      shared  buff/cache   available
Mem:           3838         382        1724         297        1731        3018
```

sdetweil, 2022-01-03:
so, i suggest u reflash the sd card and after boot, use my script to install again.. (it does all the work)

this will get you 2.18, and node 16

Fredi009sg, 2022-01-03:
> so, i suggest u reflash the sd card and after boot, use my script to install again.. (it does all the work)
> 
> this will get you 2.18, and node 16

sry but im not willing to reflash the sd card. This will delete all my other changes. Is there a way to do it with reflashing the sd card ?

Fredi009sg, 2022-01-03:
> hm.. i can't reproduce it.. reflashed buster (legacy) manually copied 2.17.1 (clone gets 2.18) changed local copy of install script to not clone, as I already had it.. install script installed node [...]

Yes its a 4gb pi 4

           total        used        free      shared  buff/cache   available
Mem:           3838        2177         419         234        1241        1505
Swap:            99          99           0

sdetweil, 2022-01-03:
i do not know.. something is messed up. I don't know what.

Fredi009sg, 2022-01-03:
can i delete Magic mirror and the rest ?

sdetweil, 2022-01-03:
sure, just delete the folder..  rm -rf ~/MagicMirror

to remove pm2 

pm2 unstartup systemd
(it will give u a command to copy/paste to execute) 
then 
 sudo npm uninstall pm2

after doing this reboot

then u can run the MM installer script again

Fredi009sg, 2022-01-03:
> sure, just delete the folder.. rm -rf ~/MagicMirror
> 
> to remove pm2
> 
> pm2 unstartup systemd (it will give u a command to copy/paste to execute) then sudo npm uninstall pm2
> 
> after doing this reboot
> 
> then u can run the MM installer script again

doing it right now. how can i check if it works ?

sdetweil, 2022-01-03:
it will start MM if u say yes to pm2.

Fredi009sg, 2022-01-03:
> it will start MM if u say yes to pm2.
> 
> otherwise (if u say no to use pm2) in the MagicMirror folder do
> 
> ```
> npm start 
> ```

your recommandation is to use pm2 ?

sdetweil, 2022-01-03:
yes, if u want it to start on boot, and keep running if it crashes, and record the output in case of problems..

all things u have to develop solutions for yourself otherwise..

Fredi009sg, 2022-01-03:
now everything i alive. i installed the remote controller. but i cant restart it. Got this

[image]

sdetweil, 2022-01-03:
the remote module  defaults to the pm2 process name is 'mm'..

my setup script does 'MagicMirror'

so u have to set it in remote module config 
pm2ProcessName: 'MagicMirror'

Fredi009sg, 2022-01-03:
did it but nothing change..

sdetweil, 2022-01-03:
so that is in the module config section, right? 
```
{
   module: 
   position: 
   config: { 
       pm2ProcessName: 'MagicMirror'
    }
}
```
the module doesn't put out any debug info.. sorry, thats all I know about that module (from looking at the code)

Fredi009sg, 2022-01-03:
/*************** AUTO GENERATED BY REMOTE CONTROL MODULE ***************/

var config = 
{
  address: '0.0.0.0',
  electronOptions: {
    fullscreen: true,
    width: 3840,
    x: 1920
  },
  ipWhitelist: [],
  language: 'de',
  modules: [
    {
      module: 'alert',
      config: {}
    },
    {
      module: 'updatenotification',
      position: 'top_bar',
      config: {}
    },
    {
      module: 'clock',
      position: 'top_left',
      config: {}
    },
    {
      module: 'calendar',
      header: 'Termine',
      position: 'top_left',
      config: {
        calendars: [
          {
            symbol: 'calendar-check',
            url: 'webcal://p12-caldav.icloud.com/published/2/MTk3NDg4NzIwOTE5NzQ4OJen2vZPDPO6ZjKDI5IIr79mZRTz4px5MNIbzMHGisQCPVn-bb7-rMKJRQXT-HvKaLs2H8xYBU6yq8si1DE_1Po'
          }
        ]
      }
    },
    {
      module: 'compliments',
      position: 'top_center',
      config: {
        compliments: {
          anytime: [
            'Na du!'
          ],
          morning: [
            'Guten morgen',
            'Genies deinen Tag!'
          ],
          afternoon: [
            'Du siehst gut aus!'
          ],
          evening: [
            'Bettzeit',
            'Schlaf gut',
            'Ab ins Bett'
          ],
          '....-01-01': [
            'Happy new year!'
          ]
        }
      },
      header: ''
    },
    {
      module: 'weather',
      position: 'top_right',
[... 168 lines cut ...]
      config: {
        success: true,
        query: {
          data: 'defaultConfig',
          module: 'MMM-MyGarbage'
        },
        data: {
          alert: false,
          weeksToDisplay: 3,
          limitTo: 99,
          dateFormat: 'dddd D MMMM',
          fade: true,
          fadePoint: 0.25
        }
      }
    }
  ],
  useHttps: false,
  httpsPrivateKey: '',
  httpsCertificate: '',
  locale: 'en-US'
}

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

Fredi009sg, 2022-01-03:
> 

You mean i have to put it into the config.js file. like i post above.

Fredi009sg, 2022-01-03:
> 

Done.. Where i have to place it ?

sdetweil, 2022-01-03:
but we are past the original error..  system is running now

sdetweil, 2022-01-03:
>Done.. Where i have to place it ?

u did it correctly. (many users don't understand config:{} is parms TO the module..)

i don't know what the module problem is.. as I said, there is no debug info produced

I have a module that restarts MM via pm2, but uses my own code, and not the library he uses.. 

here is about MagicMirror, not any modules.

Fredi009sg, 2022-01-03:
is there any way to check the hole system in futher ? where i have an eye on it ?

Fredi009sg, 2022-01-03:
Thats what i found in the error log of pm2 

/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT

what does it mean ?

sdetweil, 2022-01-03:
did u stop and restart magicmirror?  it has to kill electron to restart it  (in linux that is a signal)

Fredi009sg, 2022-01-03:
in the console with pm2 restart MagicMirror

sdetweil, 2022-01-03:
yes, so it has to kill the prior to start another... so it sends an interrupt signal (SIGINT)

Fredi009sg, 2022-01-03:
means this isn't an issue, right ?

sdetweil, 2022-01-03:
correct, this is normal

Fredi009sg, 2022-01-10:
Hi again, after a while working. Screen turns into blackscreen and i have to start the magic mirror via remote control to get back the view.

Here is the log:

PM2        | 2022-01-07T21:57:33: PM2 log: pid=32343 msg=process killed
PM2        | 2022-01-07T21:57:33: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2022-01-07T21:57:33: PM2 log: App [MagicMirror:0] online
PM2        | 2022-01-07T21:57:42: PM2 log: Stopping app:MagicMirror id:0
PM2        | 2022-01-07T21:57:42: PM2 log: pid=25113 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-07T21:57:42: PM2 log: pid=25113 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-07T21:57:42: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGINT]
PM2        | 2022-01-07T21:57:42: PM2 log: pid=25113 msg=process killed
PM2        | 2022-01-07T21:57:42: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2022-01-07T21:57:42: PM2 log: App [MagicMirror:0] online
PM2        | 2022-01-09T21:39:44: PM2 log: Stopping app:MagicMirror id:0
PM2        | 2022-01-09T21:39:44: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGINT]
PM2        | 2022-01-09T21:39:44: PM2 log: pid=25252 msg=process killed
PM2        | 2022-01-09T21:39:44: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2022-01-09T21:39:44: PM2 log: App [MagicMirror:0] online

/home/pi/.pm2/logs/MagicMirror-error.log last 15 lines:
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
0|MagicMir | [07.01.2022 21:57.40.285] [WARN]  You're using a full whitelist configuration to allow for all IPs
0|MagicMir | [07.01.2022 21:57.46.055] [WARN]  You're using a full whitelist configuration to allow for all IPs
0|MagicMir |
0|MagicMir | <--- Last few GCs --->
0|MagicMir |
0|MagicMir | [1:0x2e160000] 104344852 ms: Scavenge 879.9 (981.0) -> 879.0 (981.0) MB, 10.3 / 0.1 ms  (average mu = 0.928, current mu = 0.480) allocation failure
0|MagicMir | [1:0x2e160000] 104349947 ms: Scavenge 879.9 (981.0) -> 879.0 (982.0) MB, 6.8 / 0.0 ms  (average mu = 0.928, current mu = 0.480) allocation failure
0|MagicMir | [1:0x2e160000] 104354859 ms: Scavenge 880.0 (982.0) -> 879.1 (984.0) MB, 11.9 / 0.7 ms  (average mu = 0.928, current mu = 0.480) allocation failure
0|MagicMir |
0|MagicMir |
0|MagicMir | <--- JS stacktrace --->
0|MagicMir |
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
0|MagicMir | [09.01.2022 21:39.48.751] [WARN]  You're using a full whitelist configuration to allow for all IPs

/home/pi/.pm2/logs/MagicMirror-out.log last 15 lines:
0|MagicMir | [10.01.2022 17:46.24.490] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [10.01.2022 17:48.19.392] [INFO]  Calendar-Fetcher: Broadcasting 5 events.
0|MagicMir | [10.01.2022 17:48.20.285] [INFO]  Calendar-Fetcher: Broadcasting 2 events.
0|MagicMir | [10.01.2022 17:51.24.823] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [10.01.2022 17:53.21.458] [INFO]  Calendar-Fetcher: Broadcasting 5 events.
0|MagicMir | [10.01.2022 17:53.22.209] [INFO]  Calendar-Fetcher: Broadcasting 2 events.
0|MagicMir | [10.01.2022 17:56.25.164] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [10.01.2022 17:58.23.263] [INFO]  Calendar-Fetcher: Broadcasting 5 events.
0|MagicMir | [10.01.2022 17:58.24.087] [INFO]  Calendar-Fetcher: Broadcasting 2 events.
0|MagicMir | [10.01.2022 18:01.25.491] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [10.01.2022 18:03.25.085] [INFO]  Calendar-Fetcher: Broadcasting 5 events.
0|MagicMir | [10.01.2022 18:03.25.775] [INFO]  Calendar-Fetcher: Broadcasting 2 events.
0|MagicMir | [10.01.2022 18:06.25.823] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [10.01.2022 18:08.26.970] [INFO]  Calendar-Fetcher: Broadcasting 5 events.
0|MagicMir | [10.01.2022 18:08.27.714] [INFO]  Calendar-Fetcher: Broadcasting 2 events.

****

sdetweil, 2022-01-10:
you need to clear the logs or print more of them

the issues at the top are from Jan 7, and the stuff from below from Jan 10

do 
```
pm2 stop all
pm2 flush
pm2 start MagicMirror
```

then when it fails. pm2 stop all

pm2 logs --lines=100 >somefile.txt

ctrl-c

then examine somefile.txt

Fredi009sg, 2022-01-11:
PM2        | 2022-01-04T23:00:57: PM2 log: PM2 home             : /home/pi/.pm2
PM2        | 2022-01-04T23:00:57: PM2 log: PM2 PID file         : /home/pi/.pm2/pm2.pid
PM2        | 2022-01-04T23:00:57: PM2 log: RPC socket file      : /home/pi/.pm2/rpc.sock
PM2        | 2022-01-04T23:00:57: PM2 log: BUS socket file      : /home/pi/.pm2/pub.sock
PM2        | 2022-01-04T23:00:57: PM2 log: Application log path : /home/pi/.pm2/logs
PM2        | 2022-01-04T23:00:57: PM2 log: Worker Interval      : 30000
PM2        | 2022-01-04T23:00:57: PM2 log: Process dump file    : /home/pi/.pm2/dump.pm2
PM2        | 2022-01-04T23:00:57: PM2 log: Concurrent actions   : 2
PM2        | 2022-01-04T23:00:57: PM2 log: SIGTERM timeout      : 1600
PM2        | 2022-01-04T23:00:57: PM2 log: ===============================================================================
PM2        | 2022-01-04T23:00:57: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2022-01-04T23:00:58: PM2 log: App [MagicMirror:0] online
PM2        | 2022-01-04T23:20:43: PM2 log: Stopping app:MagicMirror id:0
PM2        | 2022-01-04T23:20:43: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:43: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:43: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:43: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:43: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:44: PM2 log: pid=1124 msg=failed to kill - retrying in 100ms
PM2        | 2022-01-04T23:20:45: PM2 log: Process with pid 1124 still alive after 1600ms, sending it SIGKILL now...
PM2        | 2022-01-04T23:20:45: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGKILL]
PM2        | 2022-01-04T23:20:45: PM2 log: pid=1124 msg=process killed
PM2        | 2022-01-04T23:20:45: PM2 log: PM2 successfully stopped
PM2        | 2022-01-04T23:21:40: PM2 log: ===============================================================================
PM2        | 2022-01-04T23:21:40: PM2 log: --- New PM2 Daemon started ----------------------------------------------------
PM2        | 2022-01-04T23:21:40: PM2 log: Time                 : Tue Jan 04 2022 23:21:40 GMT+0100 (Mitteleuropäische Normalzeit)
PM2        | 2022-01-04T23:21:40: PM2 log: PM2 version          : 5.1.2
PM2        | 2022-01-04T23:21:40: PM2 log: Node.js version      : 16.13.1
PM2        | 2022-01-04T23:21:40: PM2 log: Current arch         : arm
PM2        | 2022-01-04T23:21:40: PM2 log: PM2 home             : /home/pi/.pm2
PM2        | 2022-01-04T23:21:40: PM2 log: PM2 PID file         : /home/pi/.pm2/pm2.pid
PM2        | 2022-01-04T23:21:40: PM2 log: RPC socket file      : /home/pi/.pm2/rpc.sock
PM2        | 2022-01-04T23:21:40: PM2 log: BUS socket file      : /home/pi/.pm2/pub.sock
PM2        | 2022-01-04T23:21:40: PM2 log: Application log path : /home/pi/.pm2/logs
PM2        | 2022-01-04T23:21:40: PM2 log: Worker Interval      : 30000
PM2        | 2022-01-04T23:21:40: PM2 log: Process dump file    : /home/pi/.pm2/dump.pm2
PM2        | 2022-01-04T23:21:40: PM2 log: Concurrent actions   : 2
PM2        | 2022-01-04T23:21:40: PM2 log: SIGTERM timeout      : 1600
PM2        | 2022-01-04T23:21:40: PM2 log: ===============================================================================
PM2        | 2022-01-04T23:21:40: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2022-01-04T23:21:41: PM2 log: App [MagicMirror:0] online
PM2        | 2022-01-06T10:07:48: PM2 log: App [MagicMirror:0] exited with code [1] via signal [SIGINT]
PM2        | 2022-01-06T10:07:48: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2022-01-06T10:07:48: PM2 log: App [MagicMirror:0] online
PM2        | 2022-01-06T10:07:49: PM2 log: Stopping app:MagicMirror id:0
PM2        | 2022-01-06T10:07:49: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGINT]
PM2        | 2022-01-06T10:07:49: PM2 log: pid=18092 msg=process killed
PM2        | 2022-01-06T10:07:49: PM2 log: PM2 successfully stopped
PM2        | 2022-01-06T10:08:37: PM2 log: ===============================================================================
PM2        | 2022-01-06T10:08:37: PM2 log: --- New PM2 Daemon started ----------------------------------------------------
PM2        | 2022-01-06T10:08:37: PM2 log: Time                 : Thu Jan 06 2022 10:08:37 GMT+0100 (Mitteleuropäische Normalzeit)
PM2        | 2022-01-06T10:08:37: PM2 log: PM2 version          : 5.1.2
[... cut ...]
0|MagicMir | [11.01.2022 09:22.40.690] [INFO]  Calendar-Fetcher: Broadcasting 2 events.
0|MagicMir | [11.01.2022 09:23.47.000] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [11.01.2022 09:25.12.625] [INFO]  Calendar-Fetcher: Broadcasting 5 events.
0|MagicMir | [11.01.2022 09:27.41.232] [INFO]  Calendar-Fetcher: Broadcasting 2 events.
0|MagicMir | [11.01.2022 09:28.47.255] [INFO]  Newsfeed-Fetcher: Broadcasting 39 items.
0|MagicMir | [11.01.2022 09:30.13.906] [INFO]  Calendar-Fetcher: Broadcasting 5 events.

khassel, 2022-01-11:
googling `EAI_AGAIN`:

> EAI_AGAIN is a DNS lookup timed out error, means it is a network connectivity error or proxy related error. 

So you have problems with your internet connection, nothing we can do here.

Fredi009sg, 2022-01-12:
> googling `EAI_AGAIN`:
> 
> > EAI_AGAIN is a DNS lookup timed out error, means it is a network connectivity error or proxy related error.
> 
> So you have problems with your internet connection, nothing we can do here.

Ok can we increase the timeout maybe ? i don't know how i can solve this problem better...

sdetweil, 2022-01-12:
>Ok can we increase the timeout maybe ? i don't know how i can solve this problem better..

no, its WAY down in the system layer,   way outside MagicMirror.. 

the whole system (raspberry pi) DEPENDS on reliable network connectivity, and name lookup is how it works. 

I have seen this error a lot when the user is running a system,  like pihole, to block advertisements

Old-Engineer, 2022-03-10:
Hi,
I am having similar / the same issue.
Error is - 
[10.03.2022 03:18.39.287] [INFO]  Calendar-Fetcher: Broadcasting 10 events.
[10.03.2022 03:20.50.355] [ERROR] Calendar Error. Could not fetch calendar:  https://calendar.google.com/calendar/ical/XXXXXXXXXXXgmail.com/private-a5258b008b4d947ac79c51cc483979d6/basic.ics FetchError: request to https://calendar.google.com/calendar/ical/XXXXXXXXXXXXXgmail.com/private-a5258b008b4d947ac79c51cc483979d6/basic.ics failed, reason: getaddrinfo EAI_AGAIN calendar.google.com
    at ClientRequest.<anonymous> (/home/pi/MagicMirror/node_modules/node-fetch/lib/index.js:1461:11)
    at ClientRequest.emit (events.js:315:20)
    at TLSSocket.socketErrorListener (_http_client.js:426:9)
    at TLSSocket.emit (events.js:315:20)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  type: 'system',
  errno: 'EAI_AGAIN',
  code: 'EAI_AGAIN'
}
[10.03.2022 03:20.50.403] [ERROR] (node:28950) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'status' of undefined
    at /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:70:18
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
[10.03.2022 03:20.50.405] [ERROR] (node:28950) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
[10.03.2022 03:20.50.408] [ERROR] (node:28950) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

The pi connects wirelessly to my router.
Is there any merit in changed the DNS lookup information within the router to google's DNS servers?
It occasionally seems to ;-
a. Stop my wallpaper module from changing the background photos and/or
b. Going to a black screen. 

A Ctrl_R forces a MM reread and restarts everything but it doesn't address the issue.

Any help / advise would be appreciated

sdetweil, 2022-03-10:
EAI_AGAIN is a networking error, nothing we can do.

I have seen this many times when users are using an advertisement filter like pihole.

others have rebooted their routers.

nothing consistent. 

mm assumes a working network.  the error is reported way below any code mm provides

Old-Engineer, 2022-03-10:
@sdetweil 

Thanks

dieseltravis, 2022-05-07:
> I have seen this many times when users are using an advertisement filter like pihole.

@sdetweil I always wondered what was causing my black screen! I have to keep a keyboard attached to the pi and hit CTRL+W about once every few days. Is there a domain or URL I can add to my pihole's allow list?

sdetweil, 2022-05-07:
@dieseltravis   no idea, you would have to look at the logs to see whats going on

khassel, 2022-10-02:
no update since May, can this be closed @Fredi009sg @MichMich ?

closed 2022-10-03 by MichMich

Fredi009sg, 2023-03-03:
yes