MagicMirrorOrg/MagicMirror #2767: npm package roboto-fontface deprecated

khassel, 2022-01-04:
This is used in `fonts/package.json`.

The [Github project](https://github.com/choffmeister/roboto-fontface-bower) is archived, they recommend to use https://github.com/fontsource/fontsource instead.

I don't know if it is necassary to change this, if not feel free to close this.

rejas, 2022-01-05:
I always appreciate having maintained libaries :-) Lets see who creates a PR first :-)

khassel, 2022-01-08:
we should not work in parallel on the same issue ...

Did a first look but its not clear to me what exactly to change (beside replacing the package in `fonts/package.json`).

AFAIS the only other change would be to replace the paths/files in `fonts/roboto.css`?

rejas, 2022-01-09:
> we should not work in parallel on the same issue ...

true!
 
> Did a first look but its not clear to me what exactly to change (beside replacing the package in `fonts/package.json`).

> AFAIS the only other change would be to replace the paths/files in `fonts/roboto.css`?

Didnt have a look yet, but I assume / hope it would be like you say: 
 replace package json entry and files in the css.

khassel, 2022-01-09:
think I figured it out, will provide a PR ...

closed 2022-01-12 by khassel