MagicMirrorOrg/MagicMirror #2772: Module.createRequire is not a function, 2.18

sdetweil, 2022-01-07:
this is for documentation

if u encounter this error when doing  
```
npm run config:check 
```
after upgrading to MagicMirror 2.18, it is because the eslint package requires node 14 or above.

please check your node and nom versions by doing 
```
node -v
npm -v
```

the easiest way to upgrade node is to use the npmjs package called n (node version manager written in node js) 

do
```
sudo npm install n -g
```

then to upgrade node (and npm) do 
```
#upgrade node to version 16.1.0 as listed in MM package,json)
sudo n i 16.1.0
#fix path for this terminal session, it will be ok for new started sessions
PATH="$PATH"
```

n is pretty cool, it will let you switch between node versions too. we are only using a tiny bit of its function.

wwats, 2022-01-08:
thanks a lot. i'm a beginner and do not understand this or what i have to do exactly:

#fix path for this terminal session, it will be ok for new started sessions
PATH="$PATH"

Can You explain this? Thank You very much !

sdetweil, 2022-01-08:
you just cut and paste the text

leading # is a comment

wwats, 2022-01-08:
Thanks a lot.
My npm version is 8.1.2 - i need 8.3? i can't install because: error: eacces: permission denied, rename xxxxx
You know the solution?
My screen is still black ;-) 
i can delete this post if he is not for here... 
thanks you very much!

sdetweil, 2022-01-08:
no, whatever comes with node 16 is fine.. 

black screen is another problem .. 
see 
https://forum.magicmirror.builders/topic/15778/fix-for-black-screen-in-2-16-and-later

onisuk, 2022-01-17:
sudo n 16 gives me 

 Error: no version found for '16'

sdetweil, 2022-01-17:
what pi is this?

sdetweil, 2022-02-06:
no direct route to node 16 on pi 0w

closed 2022-04-28 by stale[bot]