MagicMirrorOrg/MagicMirror #2775: MagicMirror-error.log

andzie82, 2022-01-08:
[09.01.2022 00:12.19.217] [ERROR] (node:3205) Warning: Accessing non-existent property 'cat' of module exports inside circular dependency
(Use `electron --trace-warnings ...` to show where the warning was created)
[09.01.2022 00:12.19.218] [ERROR] (node:3205) Warning: Accessing non-existent property 'cd' of module exports inside circular dependency
[09.01.2022 00:12.19.219] [ERROR] (node:3205) Warning: Accessing non-existent property 'chmod' of module exports inside circular dependency
[09.01.2022 00:12.19.220] [ERROR] (node:3205) Warning: Accessing non-existent property 'cp' of module exports inside circular dependency
[09.01.2022 00:12.19.221] [ERROR] (node:3205) Warning: Accessing non-existent property 'dirs' of module exports inside circular dependency
[09.01.2022 00:12.19.221] [ERROR] (node:3205) Warning: Accessing non-existent property 'pushd' of module exports inside circular dependency
[09.01.2022 00:12.19.222] [ERROR] (node:3205) Warning: Accessing non-existent property 'popd' of module exports inside circular dependency
[09.01.2022 00:12.19.223] [ERROR] (node:3205) Warning: Accessing non-existent property 'echo' of module exports inside circular dependency
[09.01.2022 00:12.19.224] [ERROR] (node:3205) Warning: Accessing non-existent property 'tempdir' of module exports inside circular dependency
[09.01.2022 00:12.19.225] [ERROR] (node:3205) Warning: Accessing non-existent property 'pwd' of module exports inside circular dependency
[09.01.2022 00:12.19.225] [ERROR] (node:3205) Warning: Accessing non-existent property 'exec' of module exports inside circular dependency
[09.01.2022 00:12.19.226] [ERROR] (node:3205) Warning: Accessing non-existent property 'ls' of module exports inside circular dependency
[09.01.2022 00:12.19.227] [ERROR] (node:3205) Warning: Accessing non-existent property 'find' of module exports inside circular dependency
[09.01.2022 00:12.19.228] [ERROR] (node:3205) Warning: Accessing non-existent property 'grep' of module exports inside circular dependency
[09.01.2022 00:12.19.228] [ERROR] (node:3205) Warning: Accessing non-existent property 'head' of module exports inside circular dependency
[09.01.2022 00:12.19.229] [ERROR] (node:3205) Warning: Accessing non-existent property 'ln' of module exports inside circular dependency
[09.01.2022 00:12.19.229] [ERROR] (node:3205) Warning: Accessing non-existent property 'mkdir' of module exports inside circular dependency
[09.01.2022 00:12.19.230] [ERROR] (node:3205) Warning: Accessing non-existent property 'rm' of module exports inside circular dependency
[09.01.2022 00:12.19.231] [ERROR] (node:3205) Warning: Accessing non-existent property 'mv' of module exports inside circular dependency
[09.01.2022 00:12.19.231] [ERROR] (node:3205) Warning: Accessing non-existent property 'sed' of module exports inside circular dependency
[09.01.2022 00:12.19.232] [ERROR] (node:3205) Warning: Accessing non-existent property 'set' of module exports inside circular dependency
[09.01.2022 00:12.19.233] [ERROR] (node:3205) Warning: Accessing non-existent property 'sort' of module exports inside circular dependency
[09.01.2022 00:12.19.234] [ERROR] (node:3205) Warning: Accessing non-existent property 'tail' of module exports inside circular dependency
[09.01.2022 00:12.19.234] [ERROR] (node:3205) Warning: Accessing non-existent property 'test' of module exports inside circular dependency
[09.01.2022 00:12.19.235] [ERROR] (node:3205) Warning: Accessing non-existent property 'to' of module exports inside circular dependency
[09.01.2022 00:12.19.236] [ERROR] (node:3205) Warning: Accessing non-existent property 'toEnd' of module exports inside circular dependency
[09.01.2022 00:12.19.236] [ERROR] (node:3205) Warning: Accessing non-existent property 'touch' of module exports inside circular dependency
[09.01.2022 00:12.19.237] [ERROR] (node:3205) Warning: Accessing non-existent property 'uniq' of module exports inside circular dependency
[09.01.2022 00:12.19.237] [ERROR] (node:3205) Warning: Accessing non-existent property 'which' of module exports inside circular dependency
[09.01.2022 00:12.34.335] [ERROR] (node:3205) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

sdetweil, 2022-01-09:
that is a library used by alexacontrol. nothing we can do about it

sakshamd12, 2022-01-09:
- [Sol dty ] - - [ ] Ggu?

sdetweil, 2022-01-09:
@sakshamd12   I do not understand your last message

closed 2022-01-11 by andzie82