MagicMirrorOrg/MagicMirror #2779: Windows installation throws an error

MichlF, 2022-01-10:
I reported an issue in the installer repository for MagicMirror [here](https://github.com/sdetweil/MagicMirror_scripts/issues/42) but was asked to open an issue in this repository instead.

**Platform**: Windows 10 latest updates
**Node Version**: Node.js 16.13.1, npm 8.1.2
**MagicMirror Version**: 2.18.0

**Description**: 

I tried to install MM following the documentation on how to install on Windows (including the special steps with respect to Windows installations that all ran fine no matter whether I executed them before or after the error). However when running `npm install` inside the MagicMirror folder, tan error is thrown repeatedly at the same stage:
```
> magicmirror@2.18.0 prepare
> [ -f node_modules/.bin/husky ] && husky install || echo no husky installed.

Der Befehl "[" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
no husky installed.
```
The error message losely translates to "The command "[" is either spelled incorrectly or could not be found."
Following this error I am getting an oddly reassuring... :

```
up to date, audited 757 packages in 2s

66 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

... but magic mirror won't start:

```
C:\MM\MagicMirror>npm run start

> magicmirror@2.18.0 start
> node_modules/.bin/electron js/electron.js

Der Befehl "node_modules" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
```

sdetweil, 2022-01-11:
can do , from the MagicMirror folder
```
dir
```

MichlF, 2022-01-11:
```
C:\MM\MagicMirror>dir
 Datenträger in Laufwerk C: ist Windows
 Volumeseriennummer: 10FF-6E53

 Verzeichnis von C:\MM\MagicMirror

11.01.2022  00:29    <DIR>          .
11.01.2022  00:29    <DIR>          ..
11.01.2022  00:15               253 .editorconfig
11.01.2022  00:15                35 .eslintignore
11.01.2022  00:15               641 .eslintrc.json
11.01.2022  00:15    <DIR>          .github
11.01.2022  00:15             1,176 .gitignore
11.01.2022  00:15    <DIR>          .husky
11.01.2022  00:15                52 .prettierignore
11.01.2022  00:15                32 .prettierrc.json
11.01.2022  00:15               138 .stylelintrc.json
11.01.2022  00:15            60,252 CHANGELOG.md
11.01.2022  00:15    <DIR>          clientonly
11.01.2022  00:15    <DIR>          config
11.01.2022  00:15    <DIR>          css
11.01.2022  00:19    <DIR>          fonts
11.01.2022  00:15             2,915 index.html
11.01.2022  00:15    <DIR>          installers
11.01.2022  00:15    <DIR>          js
11.01.2022  00:15               274 jsconfig.json
11.01.2022  00:15             1,115 LICENSE.md
11.01.2022  00:29                 0 magicmirror@2.18.0
11.01.2022  00:15             1,072 module-types.ts
11.01.2022  00:15    <DIR>          modules
11.01.2022  00:25    <DIR>          node_modules
11.01.2022  00:25           548,773 package-lock.json
11.01.2022  00:20             4,540 package.json
11.01.2022  00:15             3,507 README.md
11.01.2022  00:15    <DIR>          serveronly
11.01.2022  00:15    <DIR>          splashscreen
11.01.2022  00:15    <DIR>          tests
11.01.2022  00:15    <DIR>          translations
11.01.2022  00:19    <DIR>          vendor
11.01.2022  00:29                 0 [
              17 Datei(en),        624,775 Bytes
              17 Verzeichnis(se), 329,608,699,904 Bytes frei
```

khassel, 2022-01-11:
so did `npm install` something? Is there any content in `node_modules`?

You could remove the `prepare` line in `package.json` ([this one](https://github.com/MichMich/MagicMirror/blob/a6cbc9f0ef46e4ba45e58e17f64f74410d8fe0b5/package.json#L27)) **and** delete the comma at the end of the line before and try again.

But I'm still struggling why you are the first windows user reporting this because this stuff is already there since a few releases ...

sdetweil, 2022-01-11:
so, the problem is here that the husky install fails, and electron is not yet installed (as it runs a post install script) 

SO, use 
```
npm install --only=prod
```
to get only the production stuff, not any dev dependencies

then do the fonts and vendor npm install steps 
then edit the package.json to remove DISPLAY stuff

then npm start will work .

MichlF, 2022-01-12:
Unfortunately, that does also not work for me. Apparantly because of the same issue:

```
(base) PS C:\mm> git clone https://github.com/MichMich/MagicMirror
Cloning into 'MagicMirror'...
remote: Enumerating objects: 22208, done.
remote: Total 22208 (delta 0), reused 0 (delta 0), pack-reused 22208 eceiving objects: 100% (22208/22208), 11.33 MiB | 1Receiving objects: 100% (22208/22208), 18.41 MiB | 12.28 MiB/s, done.

Resolving deltas: 100% (13598/13598), done.
(base) PS C:\mm> cd .\MagicMirror\
(base) PS C:\mm\MagicMirror> npm install --only=prod

> magicmirror@2.18.0 install
> echo "Installing vendor files ...
> " && cd vendor && npm install --loglevel=error

"Installing vendor files ...

> magicmirror@2.18.0 postinstall
> npm run install-fonts && echo "MagicMirror installation finished successfully!
> "

> magicmirror@2.18.0 install-fonts
> echo "Installing fonts ...
> " && cd fonts && npm install --loglevel=error

"Installing fonts ...
"MagicMirror installation finished successfully!

> magicmirror@2.18.0 prepare
> [ -f node_modules/.bin/husky ] && husky install || echo no husky installed.

Der Befehl "[" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
no husky installed.

added 266 packages, and audited 267 packages in 6s

19 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
(base) PS C:\mm\MagicMirror>
```

Completing the remaining steps (i.e. fonts, vendor and display stuff) doesn't help either as `npm run start` fails with with the same error as before:

```
(base) PS C:\mm\MagicMirror> cp config/config.js.sample config/config.js
(base) PS C:\mm\MagicMirror> cd fonts
(base) PS C:\mm\MagicMirror\fonts> npm install

added 1 package, and audited 2 packages in 569ms

found 0 vulnerabilities
(base) PS C:\mm\MagicMirror\fonts> cd..
(base) PS C:\mm\MagicMirror> cd vendor
(base) PS C:\mm\MagicMirror\vendor> npm install

added 9 packages, and audited 10 packages in 2s

found 0 vulnerabilities
(base) PS C:\mm\MagicMirror\vendor> cd..
(base) PS C:\mm\MagicMirror> npm run start

> magicmirror@2.18.0 start
> node_modules/.bin/electron js/electron.js

Der Befehl "node_modules" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
```

sdetweil, 2022-01-12:
what a mess

you don't have to do the fonts/vendor steps, those look ok. 

change the start line in package.json like this 
npm(16)  has broken windows exec with /  
note the double backslash 
```
"start": "node_modules\\.bin\\electron js\\electron.js",
```

MichlF, 2022-01-12:
Great, thanks that seemed to have fixed it!

khassel, 2022-01-13:
@sdetweil if this is now the standard way to start mm on windows we should update [the docs](https://github.com/MichMich/MagicMirror-Documentation/blob/master/getting-started/installation.md#windows)

sdetweil, 2022-01-13:
@khassel I've got no clue. I consider it broken.
npm stuff should be platform neutral. I proposed a PR to try to neutralize some of this last year. but as Windows isn't supported it hasn't had any interest.

besides, MagicMirror doesn't support Windows, so why is there ANY doc on Windows?

MagicMirror doesn't support pi0w either and it is now broken, needing a special install of a downlevel browser to get anything to work. 

which ditch do you want to be in, tracking all the stuff MagicMirror doesn't support, or silence on them?

closed 2022-04-28 by stale[bot]