MagicMirrorOrg/MagicMirror #2789: Electron release cadence

khassel, 2022-01-17:
With mm `v2.18.0` we moved from electron `v13` to `v16`. Every change of the electron version causes problems as .e.g new cors errors.
Main problem is not electron but the used chromium version.

@sdetweil already postet [this article](https://therecord.media/chrome-will-limit-access-to-private-networks-citing-security-reasons/amp/) in the forum which will bring new problems with next chromium/electron releases.

I found [this article](https://www.electronjs.org/blog/8-week-cadence) which describes the release frequency of electron and the supported versions.

If we don't want to run out of a supported version we have to migrate to electron `v17` before May'22 which means effectively with next mm release in April.

One good argument for not running an unsupported version was the certificate problem in October, you won't get an electron fix for such a version.

We should discuss here how we want to handle this and if we upgrade electron with the next mm release or not. In the end @MichMich has to decide this.

MichMich, 2022-01-18:
It might be a good idea to upgrade the dev branch now. If we don't run into issues coming 2 months we can release it, but we can always roll back. Agree?

rejas, 2022-01-18:
Agree

khassel, 2022-01-18:
release of `v17` is Feb'22

closed 2022-02-28 by khassel