MagicMirrorOrg/MagicMirror #2790: Magic Mirror startet nicht

Jan-2005, 2022-01-20:
`pi@dashboard:~ $ pm2 logs mm
[TAILING] Tailing last 15 lines for [mm] process (change the value with --lines option)
/home/pi/.pm2/logs/mm-error.log last 15 lines:
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm       | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP

/home/pi/.pm2/logs/mm-out.log last 15 lines:
0|mm       | [20.01.2022 20:10.19.897] [LOG]   Module helper loaded: newsfeed
0|mm       | [20.01.2022 20:10.19.898] [LOG]   All module helpers loaded.
0|mm       | [20.01.2022 20:10.20.008] [LOG]   Starting server on port 8080 ...
0|mm       | [20.01.2022 20:10.20.026] [LOG]   Server started ...
0|mm       | [20.01.2022 20:10.20.027] [LOG]   Connecting socket for: updatenotification
0|mm       | [20.01.2022 20:10.20.028] [LOG]   Starting module helper: updatenotification
0|mm       | [20.01.2022 20:10.20.029] [LOG]   Connecting socket for: calendar
0|mm       | [20.01.2022 20:10.20.030] [LOG]   Starting node helper for: calendar
0|mm       | [20.01.2022 20:10.20.030] [LOG]   Connecting socket for: newsfeed
0|mm       | [20.01.2022 20:10.20.031] [LOG]   Starting node helper for: newsfeed
0|mm       | [20.01.2022 20:10.20.031] [LOG]   Sockets connected & modules started ...
0|mm       |
0|mm       | > magicmirror@2.18.0 start
0|mm       | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|mm       |

0|mm  | [20.01.2022 20:10.22.900] [LOG]
0|mm  | Starting MagicMirror: v2.18.0
0|mm  | [20.01.2022 20:10.22.913] [LOG]
0|mm  | Loading config ...
0|mm  | [20.01.2022 20:10.22.919] [LOG]
0|mm  | Loading module helpers ...
0|mm  | [20.01.2022 20:10.22.922] [LOG]
0|mm  | No helper found for module: alert.
0|mm  | [20.01.2022 20:10.22.940] [LOG]
0|mm  | Initializing new module helper ...
0|mm  | [20.01.2022 20:10.22.941] [LOG]
0|mm  | Module helper loaded: updatenotification
0|mm  | [20.01.2022 20:10.22.944] [LOG]
0|mm  | No helper found for module: clock.
0|mm  | [20.01.2022 20:10.23.095] [LOG]
0|mm  | Initializing new module helper ...
0|mm  | [20.01.2022 20:10.23.096] [LOG]
0|mm  | Module helper loaded: calendar
0|mm  | [20.01.2022 20:10.23.099] [LOG]
0|mm  | No helper found for module: compliments.
0|mm  | [20.01.2022 20:10.23.101] [LOG]
0|mm  | No helper found for module: weather.
0|mm  | [20.01.2022 20:10.23.133] [LOG]
0|mm  | Initializing new module helper ...
0|mm  | [20.01.2022 20:10.23.134] [LOG]
0|mm  | Module helper loaded: newsfeed
0|mm  | [20.01.2022 20:10.23.135] [LOG]
0|mm  | All module helpers loaded.
0|mm  | [20.01.2022 20:10.23.241] [LOG]
0|mm  | Starting server on port 8080 ...
0|mm  | [20.01.2022 20:10.23.259] [LOG]
0|mm  | Server started ...
0|mm  | [20.01.2022 20:10.23.261] [LOG]
0|mm  | Connecting socket for: updatenotification
0|mm  | [20.01.2022 20:10.23.262] [LOG]
0|mm  | Starting module helper: updatenotification
0|mm  | [20.01.2022 20:10.23.263] [LOG]   Connecting socket for: calendar
0|mm  | [20.01.2022 20:10.23.265] [LOG]
0|mm  | Starting node helper for: calendar
0|mm  | [20.01.2022 20:10.23.266] [LOG]   Connecting socket for: newsfeed
0|mm  | [20.01.2022 20:10.23.267] [LOG]   Starting node helper for: newsfeed
0|mm  | [20.01.2022 20:10.23.268] [LOG]
0|mm  | Sockets connected & modules started ...
0|mm  | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm  | > magicmirror@2.18.0 start
0|mm  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|mm  | [20.01.2022 20:10.28.802] [LOG]
0|mm  | Starting MagicMirror: v2.18.0
0|mm  | [20.01.2022 20:10.28.815] [LOG]
0|mm  | Loading config ...
0|mm  | [20.01.2022 20:10.28.823] [LOG]
0|mm  | Loading module helpers ...
0|mm  | [20.01.2022 20:10.28.827] [LOG]
0|mm  | No helper found for module: alert.
0|mm  | [20.01.2022 20:10.28.845] [LOG]
0|mm  | Initializing new module helper ...
0|mm  | [20.01.2022 20:10.28.849] [LOG]
0|mm  | Module helper loaded: updatenotification
0|mm  | [20.01.2022 20:10.28.853] [LOG]
0|mm  | No helper found for module: clock.
0|mm  | [20.01.2022 20:10.29.004] [LOG]
0|mm  | Initializing new module helper ...
0|mm  | [20.01.2022 20:10.29.008] [LOG]
0|mm  | Module helper loaded: calendar
0|mm  | [20.01.2022 20:10.29.010] [LOG]
0|mm  | No helper found for module: compliments.
0|mm  | [20.01.2022 20:10.29.012] [LOG]
0|mm  | No helper found for module: weather.
0|mm  | [20.01.2022 20:10.29.045] [LOG]
0|mm  | Initializing new module helper ...
0|mm  | [20.01.2022 20:10.29.046] [LOG]
0|mm  | Module helper loaded: newsfeed
0|mm  | [20.01.2022 20:10.29.047] [LOG]   All module helpers loaded.
0|mm  | [20.01.2022 20:10.29.164] [LOG]
0|mm  | Starting server on port 8080 ...
0|mm  | [20.01.2022 20:10.29.181] [LOG]
0|mm  | Server started ...
0|mm  | [20.01.2022 20:10.29.183] [LOG]
0|mm  | Connecting socket for: updatenotification
0|mm  | [20.01.2022 20:10.29.185] [LOG]
0|mm  | Starting module helper: updatenotification
0|mm  | [20.01.2022 20:10.29.186] [LOG]   Connecting socket for: calendar
0|mm  | [20.01.2022 20:10.29.187] [LOG]   Starting node helper for: calendar
0|mm  | [20.01.2022 20:10.29.188] [LOG]   Connecting socket for: newsfeed
0|mm  | [20.01.2022 20:10.29.190] [LOG]   Starting node helper for: newsfeed
0|mm  | [20.01.2022 20:10.29.190] [LOG]   Sockets connected & modules started ...
0|mm  | /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
0|mm  | > magicmirror@2.18.0 start
0|mm  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
`

`pi@dashboard:~/MagicMirror $ npm run config:check

> magicmirror@2.18.0 config:check
> node js/check_config.js

[20.01.2022 20:25.24.343] [INFO]  Checking file...  /home/pi/MagicMirror/config/config.js
[20.01.2022 20:25.24.435] [INFO]  Your configuration file doesn't contain syntax errors :)
pi@dashboard:~/MagicMirror $
`

khassel, 2022-01-20:
which [raspian os](https://www.raspberrypi.com/software/operating-systems/) did you install on your sd-card?

From the error message it looks like you used the lite version (which will not work as stated in the docs).

Jan-2005, 2022-01-21:
$ cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@dashboard:~ $

 (No Lite-Version )

khassel, 2022-01-21:
can you post the output of `cat /boot/issue.txt`?

sdetweil, 2022-01-21:
@Jan-2005 is this the legacy buster or before 

show results of uname -a 

if the kernel number starts with 5 (and codename=buster), its legacy, 4 its old buster

Jan-2005, 2022-01-21:
> kannst du die Ausgabe von posten `cat /boot/issue.txt`?

pi@dashboard:~ $ cat /boot/issue.txt
Raspberry Pi reference 2021-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fa45ccf5a4b183ee566b36d74fb4b65bf9358bed, stage4

Jan-2005, 2022-01-21:
> uname -a

Linux dashboard 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

khassel, 2022-01-21:
looks all good so far, stage 4 means desktop version ([link](https://github.com/RPi-Distro/pi-gen#raspbian-stage-overview)), no idea whats wrong ...

sdetweil, 2022-01-21:
ok, legacy, on armv7 (pi3 or above) 

how did you install??

node -v 
npm -v

default node is 12.2

MM needs 14 at least, 16 prefered

sdetweil, 2022-01-21:
@khassel   issue.txt..   thanks will add that to my install  for logging

Jan-2005, 2022-01-21:
> ok, Legacy, auf armv7 (pi3 oder höher)
> 
> wie hast du installiert??
> 
> Knoten -v npm -v
> 
> Standardknoten ist 12.2
> 
> MM benötigt mindestens 14, bevorzugt 16

 npm -v
8.1.2

Knoten -v ?

khassel, 2022-01-21:
> Knoten -v ?

special tip: your pi don't speak german

sdetweil, 2022-01-21:
Knoten -v ?

was translated
```
node -v
```

Jan-2005, 2022-01-21:
> Knoten -v ?
> 
> wurde übersetzt
> 
> ```
> node -v
> ```

pi@dashboard:~ $ node -v
v16.13.2

sdetweil, 2022-01-21:
ok, i don't know.. could you try my install script? 
see https://github.com/sdetweil/MagicMirror_scripts

rename the MagicMirror folder out of the way first

```
cd ~
mv MagicMirror MagicMirror.save
```

Jan-2005, 2022-01-21:
Have already tried both, have already installed MM from scratch

sdetweil, 2022-01-21:
tried both??   there is only one

if u did not rename, then the install aborted when it found the MagicMirror folder, so as not to screw up a working installation

show me the ~/install.log (u can attach here)

Jan-2005, 2022-01-21:
``install log being saved to /home/pi/install.log
Installing helper tools ...
Updating packages ...
[                                         <=>                                                                                                                                                                                               ]
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v16.9.1
Installed Node version: v16.13.2
No Node.js upgrade necessary.
Check current NPM installation ...
NPM currently installed. Checking version number.
Minimum npm version: V7.11.2
Installed npm version: V8.3.2
No npm upgrade necessary.
Cloning MagicMirror ...
Klone nach 'MagicMirror' ...
remote: Enumerating objects: 330, done.
remote: Counting objects: 100% (330/330), done.
remote: Compressing objects: 100% (289/289), done.
remote: Total 330 (delta 51), reused 148 (delta 29), pack-reused 0
Empfange Objekte: 100% (330/330), 448.62 KiB | 5.10 MiB/s, Fertig.
Löse Unterschiede auf: 100% (51/51), Fertig.
Cloning MagicMirror Done!
Installing dependencies ...
Dependencies installation Done!
setting up initial config.js
Check plymouth installation ...
Splashscreen: Checking themes directory.
Splashscreen: Create theme directory if not exists.

Splashscreen: Changed theme to MagicMirror successfully.
Do you want use pm2 for auto starting of your MagicMirror (y/N)?y
install and setup pm2
[PM2] Applying action restartProcessId on app [MagicMirror](ids: [ 1 ])
[PM2] [MagicMirror](1) ✓
┌─────┬────────────────┬─────────────┬─────────┬─────────┬──────────┬────────┬──────┬───────────┬──────────┬──────────┬──────────┬──────────┐
│ id  │ name           │ namespace   │ version │ mode    │ pid      │ uptime │ ↺    │ status    │ cpu      │ mem      │ user     │ watching │
├─────┼────────────────┼─────────────┼─────────┼─────────┼──────────┼────────┼──────┼───────────┼──────────┼──────────┼──────────┼──────────┤
│ 1   │ MagicMirror    │ default     │ 2.18.0  │ fork    │ 1766     │ 0s     │ 179… │ online    │ 0%       │ 2.4mb    │ pi       │ enabled  │
└─────┴────────────────┴─────────────┴─────────┴─────────┴──────────┴────────┴──────┴───────────┴──────────┴──────────┴──────────┴──────────┘
[PM2] Saving current process list...
[PM2] Successfully saved in /home/pi/.pm2/dump.pm2
Do you want to disable the screen saver? (y/N)?y

We're ready! Run pm2 start MagicMirror from the ~/MagicMirror directory to start your MagicMirror.

pi@dashboard:~ $ ^C
pi@dashboard:~ $ pm2 start MagicMirror
[PM2] Applying action restartProcessId on app [MagicMirror](ids: [ 1 ])
[PM2] [MagicMirror](1) ✓
[PM2] Process successfully started
┌─────┬────────────────┬─────────────┬─────────┬─────────┬──────────┬────────┬──────┬───────────┬──────────┬──────────┬──────────┬──────────┐
│ id  │ name           │ namespace   │ version │ mode    │ pid      │ uptime │ ↺    │ status    │ cpu      │ mem      │ user     │ watching │
├─────┼────────────────┼─────────────┼─────────┼─────────┼──────────┼────────┼──────┼───────────┼──────────┼──────────┼──────────┼──────────┤
│ 1   │ MagicMirror    │ default     │ 2.18.0  │ fork    │ 5874     │ 0s     │ 179… │ online    │ 0%       │ 2.5mb    │ pi       │ disabled │
└─────┴────────────────┴─────────────┴─────────┴─────────┴──────────┴────────┴──────┴───────────┴──────────┴──────────┴──────────┴──────────┘
pi@dashboard:~ $

sdetweil, 2022-01-21:
thanks,, the number in the pm2 status line next to uptime, the circle with arrow, is the number of restarts

more than 179..

so, lets do this 
```
pm2 stop all
cd ~/magicMirror
node_modules/.bin/electron
```
should bring up a window like this 
[image]

Jan-2005, 2022-01-21:
`pi@dashboard:~/MagicMirror $ node_modules/.bin/electron

Electron 16.0.7 - Build cross platform desktop apps with JavaScript, HTML, and CSS
Usage: electron [options] [path]

A path to an Electron app may be specified. It must be one of the following:
  - index.js file.
  - Folder containing a package.json file.
  - Folder containing an index.js file.
  - .html/.htm file.
  - http://, https://, or file:// URL.

Options:
  -i, --interactive     Open a REPL to the main process.
  -r, --require         Module to preload (option can be repeated).
  -v, --version         Print the version.
  -a, --abi             Print the Node ABI version.`

Jan-2005, 2022-01-21:
nein
I have no picture on the TV

sdetweil, 2022-01-21:
are you using vnc? or ssh? or typing on pi keyboard and screen?

Jan-2005, 2022-01-21:
VNC and SSh

Der Destop kann derzeit nicht angezeigt werden

sdetweil, 2022-01-21:
see this 

https://www.tomshardware.com/how-to/fix-cannot-currently-show-desktop-error-raspberry-pi

if u are booted to command mode, ssh would work, vnc not

Jan-2005, 2022-01-21:
i have reboot and he run

Jan-2005, 2022-01-21:
I  love You!

Jan-2005, 2022-01-21:
`pi@dashboard:~ $ pm2 logs MagicMirror
[TAILING] Tailing last 15 lines for [MagicMirror] process (change the value with                                      --lines option)
/home/pi/.pm2/logs/MagicMirror-error.log last 15 lines:
0|MagicMir | bash: /home/pi/MagicMirror/installers/mm.sh: Datei oder Verzeichnis                                      nicht gefunden
0|MagicMir | bash: /home/pi/MagicMirror/installers/mm.sh: Datei oder Verzeichnis                                      nicht gefunden
0|MagicMir | bash: /home/pi/MagicMirror/installers/mm.sh: Datei oder Verzeichnis                                      nicht gefunden
0|MagicMir | bash: /home/pi/MagicMirror/installers/mm.sh: Datei oder Verzeichnis                                      nicht gefunden
0|MagicMir | bash: /home/pi/MagicMirror/installers/mm.sh: Datei oder Verzeichnis                                      nicht gefunden
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP
0|MagicMir | /home/pi/MagicMirror/node_modules/electron/dist/electron exited wit                                     h signal SIGTRAP

/home/pi/.pm2/logs/MagicMirror-out.log last 15 lines:
0|MagicMir | [21.01.2022 22:09.07.928] [LOG]   Connecting socket for: newsfeed
0|MagicMir | [21.01.2022 22:09.07.929] [LOG]   Starting node helper for: newsfee                                     d
0|MagicMir | [21.01.2022 22:09.07.930] [LOG]   Sockets connected & modules start                                     ed ...
0|MagicMir | [21.01.2022 22:09.09.438] [LOG]   Launching application.
0|MagicMir | [21.01.2022 22:09.12.911] [LOG]   Create new calendarfetcher for ur                                     l: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval:                                      300000
0|MagicMir | [21.01.2022 22:09.12.982] [LOG]   Create new newsfetcher for url: h                                     ttps://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
0|MagicMir | [21.01.2022 22:09.13.020] [INFO]  Checking git for module: default
0|MagicMir | [21.01.2022 22:09.13.681] [INFO]  Newsfeed-Fetcher: Broadcasting 60                                      items.
0|MagicMir | [21.01.2022 22:09.14.084] [INFO]  Calendar-Fetcher: Broadcasting 10                                      events.
0|MagicMir | [21.01.2022 22:14.30.293] [INFO]  Newsfeed-Fetcher: Broadcasting 60                                      items.
0|MagicMir | [21.01.2022 22:14.31.028] [INFO]  Calendar-Fetcher: Broadcasting 10                                      events.
0|MagicMir | [21.01.2022 22:19.30.735] [INFO]  Newsfeed-Fetcher: Broadcasting 60                                      items.
0|MagicMir | [21.01.2022 22:19.31.758] [INFO]  Calendar-Fetcher: Broadcasting 10                                      events.
0|MagicMir | [21.01.2022 22:24.31.025] [INFO]  Newsfeed-Fetcher: Broadcasting 60                                      items.
0|MagicMir | [21.01.2022 22:24.32.495] [INFO]  Calendar-Fetcher: Broadcasting 10                                      events.
`

Jan-2005, 2022-01-21:
it runs but still gets error

sdetweil, 2022-01-21:
pm2 flush 
to get rid of old stuff

sdetweil, 2022-01-21:
flush old messages

sdetweil, 2022-01-21:
I have added  checks for lite mode(wrong OS image) , and running in text mode on full desktop (this problem)

sdetweil, 2022-01-24:
@Jan-2005 can you close this?

closed 2022-01-28 by Jan-2005