MagicMirrorOrg/MagicMirror #2792: Can't See Calendar, New Feed and To Do List Module ;(

lina-ja, 2022-01-21:
I Can't See Calendar, News Feed and To-Do List Module.
There are a link of these modules below.

[](https://docs.magicmirror.builders/modules/newsfeed.html)
[](https://docs.magicmirror.builders/modules/calendar.html)
[](https://github.com/cbrooker/MMM-Todoist)

This is the message from terminal.

`21.01.2022 14:49.20.550] [WARN]  You're using a full whitelist configuration to allow for all IPs
[21.01.2022 14:49.20.561] [LOG]   Server started ...
[21.01.2022 14:49.20.563] [LOG]   Connecting socket for: calendar
[21.01.2022 14:49.20.565] [LOG]   Starting node helper for: calendar
[21.01.2022 14:49.20.566] [LOG]   Connecting socket for: MMM-FastNotes
[21.01.2022 14:49.20.567] [LOG]   Starting module helper: MMM-FastNotes
[21.01.2022 14:49.20.568] [LOG]   Connecting socket for: newsfeed
[21.01.2022 14:49.20.569] [LOG]   Starting node helper for: newsfeed
[21.01.2022 14:49.20.570] [LOG]   Connecting socket for: MMM-Remote-Control
[21.01.2022 14:49.20.571] [LOG]   Starting node helper for: MMM-Remote-Control
[21.01.2022 14:49.20.579] [LOG]   Connecting socket for: MMM-Todoist
[21.01.2022 14:49.20.580] [LOG]   Starting node helper for: MMM-Todoist
[21.01.2022 14:49.20.581] [LOG]   Connecting socket for: MMM-GooglePhotos
[21.01.2022 14:49.20.582] [LOG]   Connecting socket for: MMM-awesome-alexa
[21.01.2022 14:49.20.583] [LOG]   Sockets connected & modules started ...
[21.01.2022 14:49.20.584] [LOG]   
Ready to go! Please point your browser to: http://192.168.11.10:8080
[21.01.2022 14:49.43.000] [LOG]   Create new calendarfetcher for URL: [HIDE] - Interval: 300000
[21.01.2022 14:49.43.096] [LOG]   Create new newsfetcher for url: https://news.yahoo.co.jp/rss/topics/it.xml - Interval: 300000
[21.01.2022 14:49.43.437] [INFO]  Newsfeed-Fetcher: Broadcasting 8 items.
[21.01.2022 14:49.44.483] [INFO]  Calendar-Fetcher: Broadcasting 7 events.
[21.01.2022 14:54.43.592] [INFO]  Newsfeed-Fetcher: Broadcasting 8 items.
[21.01.2022 14:54.45.867] [INFO]  Calendar-Fetcher: Broadcasting 7 events.
[21.01.2022 14:59.43.727] [INFO]  Newsfeed-Fetcher: Broadcasting 8 items.
[21.01.2022 14:59.47.143] [INFO]  Calendar-Fetcher: Broadcasting 7 events.
`
I hide the URL of the calendar but I checked this link is shared successfully.
These modules were shown before 2 days ago.
But just changed the place and connected to another Wi-Fi these are disappeared.
I didn't change anything.

But other modules are working as usual!! Exp:
( FastNote,[](https://github.com/polarbearjngl/MMM-FastNotes)
[](https://github.com/aneaville/MMM-GooglePhotos)
[](https://docs.magicmirror.builders/modules/weather.html)

Plz Help Me and give me any advice...
I already updated the version of Magic Mirror.
I have no idea now...

lina-ja, 2022-01-21:
[image]
[image]

jboucly, 2022-02-16:
Please give your `config.js`

lina-ja, 2022-02-17:
Same as [here](https://forum.magicmirror.builders/topic/16231/can-t-see-calendar-new-feed-and-to-do-list-module/13), I couldn’t solve this, so I restored it from backup.
Thank you for your corporation and Your kindness!!!

closed 2022-02-17 by lina-ja