MagicMirrorOrg/MagicMirror #2801: problmem when i run npm start

Hamzacos, 2022-01-28:
> magicmirror@2.18.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

/home/pi/MagicMirror/node_modules/electron/index.js:17
    throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again');
    ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at getElectronPath (/home/pi/MagicMirror/node_modules/electron/index.js:17:11)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/index.js:21:18)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/cli.js:3:18)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.18.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.18.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-28T20_09_06_161Z-debug.log

sdetweil, 2022-01-28:
what pi? 
uname - a 
will help

what pi os?
lsb_release -a
will help

if you used my scripted install it probably would have fixed the problem
see
https://github.com/sdetweil/MagicMirror_scripts

jboucly, 2022-02-16:
This problem is due to improper installation of electron. 

Try to remove `node_modules` and `npm cache clean --force `

After doing this, run `npm i` 

A piece of advice have a good internet connection to avoid having a corrupt installation of electron !

sdetweil, 2022-03-27:
@Hamzacos  did you resolve this?

rejas, 2022-10-02:
Maybe we shoudl close this @MichMich since @Hamzacos hasnt responded in quite a few months...

closed 2022-10-03 by MichMich