MagicMirrorOrg/MagicMirror #2803: 102730091 ms: Mark-sweep 881.5 (983.0) -> 880.4 (983.3) MB, 6928.0 / 0.5 ms  (average mu = 0.346, current mu = 0.311) allocation failure scavenge might not succeed

Fredi009sg, 2022-01-31:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the corresponding repository:

- karsten13/magicmirror: [https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)
- (deprecated) bastilimbach/docker-magicmirror: [https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 14 or later (recommended is 16).

**MagicMirror Version**: Please let us know which version of MagicMirror you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rejas, 2022-01-31:
where did you get this message and why is that a problem?

Fredi009sg, 2022-01-31:
got this when i'm doing pm2 logs

sdetweil, 2022-01-31:
@Fredi009sg 

can you do what the issue template asked for

what pi
when did you download the PI OS?

uname -a
lsb_release -a
node -v
npm -v
free -m
df -k
 
how did you install?  manual, automated script? 
 
what did you do BEFORE u attempted install?
   any system updates, or installs of other things?

Fredi009sg, 2022-02-01:
> @Fredi009sg
> 
> can you do what the issue template asked for
> 
> what pi when did you download the PI OS?
> 
> uname -a 

Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

> lsb_release -a 

No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

> node -v 

v16.13.2

> npm -v 

8.1.2

> free -m 

                      total        used        free      shared  buff/cache   available
Mem:          3838       1399         242         205        2196        2300
Swap:          1023         125         898

> df -k

Dateisystem                                     1K-Blöcke    Benutzt Verfügbar Verw% Eingehängt auf
/dev/root                                        30358348   13121948  15951904   46% /
devtmpfs                                          1800564          0   1800564    0% /dev
tmpfs                                             1965428      17220   1948208    1% /dev/shm
tmpfs                                             1965428      17060   1948368    1% /run
tmpfs                                                5120          8      5112    1% /run/lock
tmpfs                                             1965428          0   1965428    0% /sys/fs/cgroup
/dev/mmcblk0p1                                     258095      49283    208812   20% /boot
192.168.160.70:/volume1/Backups/Full_Backup_Pi 1917962752 1003842816 914001152   53% /backup
tmpfs                                              393084          4    393080    1% /run/user/1000

> how did you install? manual, automated script?

with your script..

> what did you do BEFORE u attempted install? any system updates, or installs of other things?

jboucly, 2022-02-16:
I had this problem when I created my module... 
I hadn't anticipated the available memory allocation of an RPI.

I'm guessing you've either developed a module that takes more memory allocation than the RPI suggests. Either you used a module that was not very well developed due to the memory size of an RPI. 

Can you list the modules you use ?

closed 2022-04-27 by stale[bot]