MagicMirrorOrg/MagicMirror #2804: Update da.json to v2.18.0

The-Exterminator, 2022-01-31:
```{
	"LOADING": "Indlæser &hellip;",

	"TODAY": "I dag",
	"TOMORROW": "I morgen",
	"DAYAFTERTOMORROW": "I overmorgen",
	"RUNNING": "Slutter om",
	"EMPTY": "Ingen kommende begivenheder.",
	"WEEK": "Uge {weekNumber}",

	"N": "N",
	"NNE": "NNØ",
	"NE": "NØ",
	"ENE": "ØNØ",
	"E": "Ø",
	"ESE": "ØSØ",
	"SE": "SØ",
	"SSE": "SSØ",
	"S": "S",
	"SSW": "SSV",
	"SW": "SV",
	"WSW": "VSV",
	"W": "V",
	"WNW": "VNV",
	"NW": "NV",
	"NNW": "NNV",

	"FEELS": "Føles som {DEGREE}",
	"PRECIP": "PoP",

	"MODULE_CONFIG_CHANGED": "Konfigurationsmulighederne for {MODULE_NAME} modulet er ændret.\nSe venligst dokumentationen.",
	"MODULE_CONFIG_ERROR": "Fejl i {MODULE_NAME} modulet. {ERROR}",
	"MODULE_ERROR_MALFORMED_URL": "Forkert url.",
	"MODULE_ERROR_NO_CONNECTION": "Ingen internetforbindelse.",
	"MODULE_ERROR_UNAUTHORIZED": "Godkendelse mislykkedes.",
	"MODULE_ERROR_UNSPECIFIED": "Tjek logfiler for flere detaljer.",

	"NEWSFEED_NO_ITEMS": "Ingen nyheder i øjeblikket.",

	"UPDATE_NOTIFICATION": "MagicMirror² opdatering tilgængelig.",
	"UPDATE_NOTIFICATION_MODULE": "Opdatering tilgængelig for {MODULE_NAME} modulet.",
	"UPDATE_INFO_SINGLE": "Den nuværende installation er {COMMIT_COUNT} commit bagud på {BRANCH_NAME} branch'en.",
	"UPDATE_INFO_MULTIPLE": "Den nuværende installation er {COMMIT_COUNT} commits bagud på {BRANCH_NAME} branch'en."
}
```

rejas, 2022-01-31:
I assume this should be better a PR?

The-Exterminator, 2022-01-31:
@rejas arhhhhh sorry. i will get it to pr.

The-Exterminator, 2022-01-31:
Is on [PR](https://github.com/MichMich/MagicMirror/pull/2805)

closed 2022-01-31 by The-Exterminator

rejas, 2022-01-31:
> @rejas arhhhhh sorry. i will get it to pr.

No worries :-)