MagicMirrorOrg/MagicMirror #2813: Calendar: Repeating events with time are not displayed

Psycho0verload, 2022-02-17:
# Calendar: Repeating events with time are not displayed
## Enviroment
- Raspberry Pi 3
- Raspbian GNU/Linux 11 (bullseye)
- PM2 5.1.2
- npm 8.3.1
- node v16.14.0
- MM 2.18.0
- Calendar selfhostet Nextcloud

## Problem:

- ❌ A repeating event with a fixed time is not displayed in the calendar.

## Working:

- ✅ Repeat of the full day event is displayed
- ✅ A non-repeating event with a fixed time is displayed
- ✅ A non-repeating full day event is displayed

## Configuration 
```
{
    module: "calendar",
    header: "Tagesübersicht",
    classes: "daily-overview",
    position: "middle_center",
    config: {
        carouselId: "1",
        tableClass: "large daily-overview-table wachenaufgaben",
        fetchInterval: 600000, /* 600000 */
        maxTitleLength: 50,
        displaySymbol: true,
        showLocation: false,
        maxTitleLines: 1,
        maximumEntries: 9,
        maximumNumberOfDays: 1,
        calendars: [
            {
                url: 'https://mydomain.it/remote.php/dav/public-calendars/ZMGgBLGxxxxxDw?export',
                symbolClass: 'todo-symbol',
                titleClass: 'todo-title',
                timeClass: 'todotime',
                symbol: 'clipboard-list-check'
            },
            {
                url: 'https://mydomain.it/remote.php/dav/public-calendars/WW49cKGxxxxxq8?export',
                symbolClass: 'clean-symbol',
                titleClass: 'clean-title',
                timeClass: 'clean-time',
                symbol: 'vacuum',
            },
            {
                url: 'https://mydomain.it/remote.php/dav/public-calendars/eMQW6xxxxxbXq?export',
                symbolClass: 'info-symbol',
                titleClass: 'info-title',
                timeClass: 'info-time',
                symbol: 'circle-info',
                maximumNumberOfDays: 7,
            }
        ]
    }
},
```

## Expected Results
It would be desirable to display repeating events with time as well as it works with all day events.

## Actual Results
The event disappears completely as soon as it is no longer an all-day event.

## Bug Hunt
https://forum.magicmirror.builders/topic/16372/calendar-repeating-events-with-time-are-not-displayed

Psycho0verload, 2022-02-17:
It also makes no difference whether I enter it in the web interface of nextcloud or in the calendar app from apple.

sdetweil, 2022-03-16:
@Psycho0verload the event is inside the relative window, so shown as tomorrow or Thursday

change relative  and look at urgency

sdetweil, 2022-03-27:
@Psycho0verload were u able to resolve this?

Psycho0verload, 2022-03-28:
I will check this again. I will have access to the system again tomorrow

Psycho0verload, 2022-03-28:
What do you mean? So at what point should I change something?

It is difficult to change something in the source code of the appointment itself, because the end user is not able to do that.

I have now tested the whole thing again with:
`timeFormat: absolute / relative / dateheaders`

In any case the date is displayed correctly as long as it is not repeated.

I don't see any more value to set the option `urgency`, because in every display whether absolute, relative or dateheaders the error occurs.

sdetweil, 2022-07-21:
@Psycho0verload 

looking back over this., with the failing cal entry above, 

```
RRULE:FREQ=WEEKLY;INTERVAL=1
   DTEND;TZID=Europe/Berlin:20220217T190000  // indented to align date/time for discussion
DTSTART;TZID=Europe/Berlin:20220217T090000
```
it starts and ends on the same day, and requests weekly... so would only show one event.

khassel, 2022-10-02:
seems not to be a bug, can this be closed @sdetweil ?

sdetweil, 2022-10-03:
yes, can close

khassel, 2022-10-03:
please close @Psycho0verload @MichMich

closed 2022-10-03 by Psycho0verload