MagicMirrorOrg/MagicMirror #2821: Install error: Playwright does not support firefox on ubuntu18.04-arm64

MMM-Jasmin, 2022-03-14:
Hi,

i am trying to install the newest version on an arm64 system.
Since the last version playwright was added as it seems.
When I try to execute the npm install I receive the the error:

ERROR: Playwright does not support firefox on ubuntu18.04-arm64
even firefox is not installed on the system.

any ideas?
[2022-03-14T11_57_07_251Z-debug-0.log](https://github.com/MichMich/MagicMirror/files/8244505/2022-03-14T11_57_07_251Z-debug-0.log)

sdetweil, 2022-03-14:
playwright is for testing.

do 

npm install only=prod

khassel, 2022-03-14:
as addition to @sdetweil's answer: There is already a [PR in the documention repository](https://github.com/MichMich/MagicMirror-Documentation/pull/105) reflecting this which will be hopefully merged soon.

Only users who wants to run the tests needs the devDependencies (including playwright), so for 99% of the installations running `npm install only=prod` (or `npm install omit=dev` depending on the npm version) is the better choice. Its quicker and needs less space on disk.

MichMich, 2022-03-14:
@khassel merged it!

MMM-Jasmin, 2022-03-15:
Cool thanks for the clarification! It works with --omit=dev.
But i guess it is "--only=prod" or "--omit=dev", isn´t it? They documentation now stats both

"Install the application: npm install --only=prod --omit=dev"

sdetweil, 2022-03-15:
I have updated the scripts, install and upgrade to handle the different options for prod only/omit dev

https://github.com/sdetweil/MagicMirror_scripts

khassel, 2022-03-15:
Tested now `npm install --only=prod --omit=dev` with node v16 and v17 and this works. I think the contributor of the documentation PR testet this for older versions too.

The `--omit` was introduced with npm 7.x (Node v15) so we can remove the `--only` option with end of life of Node v14 (2023-04-30).

sdetweil, 2022-03-15:
I changed my scripts to use --omit for npm 8

khassel, 2022-10-01:
should be closed @MMM-Jasmin @MichMich

closed 2022-10-06 by rejas