MagicMirrorOrg/MagicMirror #2823: on forum can we get reputation and anti spam installed.

sdetweil, 2022-03-18:
see #2726

MichMich, 2022-03-19:
Unfortunately I couldn't find any working (maintained) reputation plugins, but I did add some additional anti spam checks. Especially for the creation of new accounts. Let me know if this helps.

sdetweil, 2022-03-20:
no help, sorry.

sdetweil, 2022-04-04:
can e get spam-be-gone installed and configured?
https://community.nodebb.org/topic/15119/spam-registration-amount-and-handling-is-unbearable

still supported it says. 

 we get 20-30 new spam registrations a day

MichMich, 2022-04-05:
Spam-be-gone is already installed an active (at is has been for the past few months). 

Not sure how to catch more spam.

[image]

[image]

[image]

[image]

sdetweil, 2022-04-05:
can u up the reputation required for about me and signature to 2?

[image]

MichMich, 2022-04-05:
Good catch! Done!

sdetweil, 2022-04-05:
thx.  

can u do that for the website field?

MichMich, 2022-04-05:
Yeah, I did it for all the fields that can be abused for spam.

sdetweil, 2022-04-05:
thanks... can't wait to see how that affects registrations..

sdetweil, 2022-04-18:
so, we still get a bunch of advert registration attempts, but looks like it's trending lower.

closed 2022-04-18 by sdetweil