MagicMirrorOrg/MagicMirror #2824: node-ical update to minimum version to 0.15.0

kuosman, 2022-03-21:
Hi,
Calendar module uses node-ical. There is one issue on Google calendar and Microsoft ism.
See more: https://github.com/jens-maus/node-ical/issues/169

They have fixed this to node-ical 0.15.0 version https://github.com/jens-maus/node-ical/pull/170 & https://github.com/jens-maus/node-ical/commits/0.15.0

Maybe MM can update this also so more ics calendars are working again :)

rejas, 2022-03-25:
IIRC there was already an attempt to upgrade node-ical to a newer version which failed due to missing luxon package: https://github.com/MichMich/MagicMirror/pull/2716

Doesnt seem fixed with 0.15.0 as far as I can see. Or is that "--no-optional" install switch not relevant for us anymore @khassel ?

khassel, 2022-03-26:
This is related to #2718 

I need `--no-optional` when building docker images for server only (without electron).

Will tests this when I'm back home next week.

khassel, 2022-04-01:
this is fixed in the new v2.19.0 release and should be closed @kuosman @MichMich

closed 2022-04-02 by MichMich