MagicMirrorOrg/MagicMirror #2826: MagicMirror will not start

ejmorgan55, 2022-03-22:
MagicMirror will not start. I have run MM for over 3 years. I ran the recent package; however it will not start. I checked and upgraded Node and NPM. Still no start. Following error was given for npm start:

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.18.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

/home/pi/MagicMirror/node_modules/electron/dist/electron: error while loading shared libraries: libxkbcommon.so.0: cannot open shared object file: No such file or directory

Any ideas?

sdetweil, 2022-03-22:
you are missing a system library

sounds like you are not using the desktop

ejmorgan55, 2022-03-23:
Actually; I am using the desktop? I have the full GUI up and running without any noted problems (I sent you the error message above via chrome from the Raspberry pi desk top). Where might I find the library to download and where does it need to be placed? Very odd as I have operated MM for several years and have downloaded numerous MM systems for various projects and have never had this problem before; albeit, it sounds like an error in the OS rather than the MM.

sdetweil, 2022-03-23:
Google search.  did u load the 64 bit version?

ejmorgan55, 2022-03-23:
Fixed it; indeed I was missing the library. Not sure why. I used sudo apt-get install libxkbcommon0, and the MM came up and started working. Probably something in the OS that was not downloaded. Thanks.

sdetweil, 2022-03-27:
@ejmorgan55 can u close this now?

closed 2022-04-05 by MichMich