MagicMirrorOrg/MagicMirror #2828: HTML code in alert and notification messages is not displayed correctly

lrabius, 2022-03-26:
**Platform**: RaspberryPi 3 with Debian 10 Buster

**Node Version**: 16.14.2

**MagicMirror Version**: 2.18.0

**Description**: I wanted to show an alert with a HTML code message. The problem occurs also with notifications.

**Steps to Reproduce**: Show alert or notification with custom HTML code in the message.

**Expected Results**: The HTML code in the message should be parsed and the message should be displayed accordingly to the specified HTML code.

**Actual Results**: The complete message (HTML code and actual message) is escaped and the code and message in the alert popup is displayed as plain text.

**Additional Notes**: When you set the option `autoescape: false` in the file `js/module.js` where the nunjucks environment is created, everything works as expected and the message won't be escaped and is displayed correctly according to the provided HTML code.
But maybe there is another way to get the HTML code in alerts working again because if autoescape is deactivated this could be a security risk because code injection is possible then.

khassel, 2022-10-02:
no update since April, can this be closed? @lrabius @MichMich

grenagit, 2023-01-20:
I have the **same problem** with my module [MMM-VigilanceMeteoFrance](https://github.com/grenagit/MMM-VigilanceMeteoFrance) to display information in **bold** in a notification. HTML code is displayed in plain text with `<strong>` and `</strong>`...

I **solved** this issue with the nunjucks **safe filter** into [notification template (line 8)](https://github.com/MichMich/MagicMirror/blob/2eaf9dfeeb5a10d677bda61bf468d3ff2bd4af33/modules/default/alert/templates/notification.njk#L8):
```
  <span class="light bright small">{{ message | safe }}</span>
```

Can this patch **be integrated** into the [alert module](https://github.com/MichMich/MagicMirror/tree/master/modules/default/alert) or is the HTML code to **be banned** from notifications in MagicMirror?

khassel, 2023-01-20:
we have the `safe` statement already in other default module njk's, so I think this change is o.k.

Same should be done in [alert template](https://github.com/MichMich/MagicMirror/blob/develop/modules/default/alert/templates/alert.njk#L17).

closed 2023-04-04 by khassel