MagicMirrorOrg/MagicMirror #2831: electron app.disableHardwareAcceleration()

khassel, 2022-03-28:
Electron is throwing gpu errors under some circumstances e.g. running in a docker container as mentioned [here](https://github.com/MichMich/MagicMirror/issues/2656#issuecomment-981160715) and there were already some threads in the forum about this.

To disable these errors you can add `app.disableHardwareAcceleration()` in `electron.js` or `--disable-gpu` in `package.json`.

Both variants are ugly because for automating this you can only use `sed` or similar tools.

So I would like to add an environment variable `ELECTRON_DISABLE_GPU` and add `app.disableHardwareAcceleration()` in `electron.js` only when set.

More info's to disableHardwareAcceleration:
- https://www.electronjs.org/docs/latest/tutorial/offscreen-rendering
- https://www.electronjs.org/docs/latest/api/app#appdisablehardwareacceleration

May it is also an option to run always with `app.disableHardwareAcceleration()`?

shin10, 2022-03-29:
Have you considered to use
```js
electronSwitches: ["disable-gpu"],
```
in your `config.js`?

I feel your `sed` pain, since I have several hardware components attached to GPIO but am developing in WSL mostly. So I have to en/disable the components depending on the platform. To do this I'm running a make recipe which substitutes via `envsubst` variables sourced from an `.env` file before starting the app. That's ok-ish I guess. At least I can handle all my secrets that way, too.
It's work in progress but you can have a look at my conf [here](https://github.com/shin10/MagicMirror-config).

Anyway, I'm super new to MM so if this is a stupid idea, just ignore it. I'm open to suggestions as well.

khassel, 2022-03-29:
> electronSwitches: ["disable-gpu"],

yes, tested this with no success, `--disable-gpu` is missing in the [list of supported command line switches](https://www.electronjs.org/de/docs/latest/api/command-line-switches).

I'm already using the `envsubst` construction in the [docker entrypoint script](https://gitlab.com/khassel/magicmirror/-/blob/master/build/entrypoint.sh#L39) but thats no solution for this problem.

shin10, 2022-03-29:
Ah, I see. Well, it's been a while, but I once used software rendering for tests of pages with three.js in docker. Back then I used Chrome with OSMesa. There seems to be a similar approach for electron with SwiftShader https://medium.com/@mohamedmansour/how-to-run-electron-js-in-software-rendering-headless-49601b87961e
Maybe that's an alternative to disabling the GPU.

commit 00a7c6b5 (2022-04-01): added new env var `ELECTRON_DISABLE_GPU` which disable gpu under electron if set (fixes #2831).