MagicMirrorOrg/MagicMirror #2833: discord user reports unable to register on forum

sdetweil, 2022-03-30:
Hey folks, I am having an issue registering with your forum ( https://forum.magicmirror.builders/register )
Despite completing the capcha fields, it still thinks I haven't.
Is this a known problem, or is it just me?

MichMich, 2022-03-30:
I tried it and manage to successfully register a new account in one go. Maybe try a different browser?

rjclocks, 2022-03-30:
That's odd. 
I tried Chrome and Firefox on my PC and then Safari on my phone and had the same error each time.

rjclocks, 2022-03-30:
[image]

MichMich, 2022-03-30:
The weird thing is that I don't even see that second captcha.

rjclocks, 2022-03-30:
Somehow they are conflicting

MichMich, 2022-03-30:
Tried to disable hcaptcha but it won't let me from my phone. Will do it tomorrow morning when I'm back behind my computer.

MichMich, 2022-03-31:
Could you give it another try?

rjclocks, 2022-03-31:
Super. It worked without a problem.
Thanks you for resolving it.

MichMich, 2022-03-31:
Thanks for reporting back!