MagicMirrorOrg/MagicMirror #2835: Bad font for non latin users

alexey-gamov, 2022-04-04:
## Fonts bug in MagicMirror

**Platform**: MagicMirrorOS on Raspberry Pi 2 .
**MagicMirror² Version**: 2.19.0

**Description**:

In this update fonts package was changed from `roboto-fontface-bower` to `fontsource` (2dcf55ea89f752390554d655ee819148c07e4fbc).
CSS file `roboto.css` uses only `latin` subset for woff files = I thing that fonts must consist all world chars.

**Way to fix**:

open-sans-all-300-normal.woff

**Expected Results**: Fonts are still great with non US characters.
**Actual Results**: Fonts are ugly. Need to add custom `@font-face` from Google.

khassel, 2022-04-04:
> It is not fair = fonts must consist all world chars.

that was not intentional

Can you provide a PR to fix this? Otherwise I will create one ...

alexey-gamov, 2022-04-04:
> > It is not fair = fonts must consist all world chars.
> 
> that was not intentional
> 
> Can you provide a PR to fix this? Otherwise I will create one ...

Last package was international :)
Unfortunately am not expert in node modules, it will be better if you can fix that issue..

outlying, 2022-04-05:
I confirm that, 2.19.0 update "broke" fonts in news module and greeting module (see `ń` character)
[image]

MichMich, 2022-04-06:
Can anyone confirm the PR of @khassel solves the issue?

DecliningLotus, 2022-04-11:
Fontsource maintainer here. It may be better for you to reference the source CSS file itself so you can write the appropriate `@font-face` declarations instead to support international users. The `unicode-range` CSS selector is a key thing that @khassel’s PR is missing that prevents `woff2` files from being used effectively.

Example [CSS file](https://github.com/fontsource/fontsource/blob/main/fonts/google/roboto/400.css).

Additionally when [fontsource/fontsource#234](https://github.com/fontsource/fontsource/issues/234) is implemented, I think that will help simplify your build process a lot more. Feel free to keep tabs on that issue!

rejas, 2022-04-27:
> Can anyone confirm the PR of @khassel solves the issue?

I can confirm that with his PR the ń looks better (when used in the compliments module) Not sure if that solves it for ALL possible letters.

PR #2836 (2022-04-09): Fix problems with non latin fonds caused by updating to fontsource
fixes #2835
commit b4e75f68 (2022-04-09): fix problems with non latin fonds caused by updating to fontsource (fixes #2835).
closed 2022-07-02 by MichMich