MagicMirrorOrg/MagicMirror #2837: Calandar, stoped showing....worked last 2 years ...

BGH-H, 2022-04-11:
Hi guys, the standard MM calendar module stopped working the about a week ago. I really don't know which updates or modules have changed. Can anyone help ? Its been aged since I installed it and im not really into the matter right know to understand what's wrong. 

* Using a shared calendar iPhones with webcal file. A copy of config.sys isn't needed coz nothing changed. The mirror is running, no other errors since 2 years, no update notifications in magic mirror. We update our iPhones frequently. So it must be something in the iPhone update? Thanks in advance for looking:

sdetweil, 2022-04-11:
who is your calendar source? gmail, ???   

some of the providers are updating their access urls.. GMAIL did that

BGH-H, 2022-04-12:
Its iPhone, so no its not gmail. standard iPhone calander

sdetweil, 2022-04-12:
i would check by getting a new access url

BGH-H, 2022-04-12:
I did I created a new one and it doesn't work anymore.....neither of them... creating a public calendar and finding URL is quite simple... it just stopped working... Has any changed in regards to support of ical ? Super weird... it just stopped after working for years :(

modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "Kalender",
			position: "top_left",
			config:{
			maximumEntries: 5,
			maxTitleLines: 20,
			calendars: [
			{
				symbol: "My Calendar",
//				url: "webcal://p59-caldav.icloud.com/published/2/MTM1Njc5ODIwMDEzNTY3OS33Gnox-W4iauVmHlxciBaLip9TGlgK5cUsslutIFId999hJjMnYAwPNhAkeJ9rKj3G7Jr5p-vDAJTCXXXXXXX"
				url: "webcal://p53-caldav.icloud.com/published/2/MTc4ODM5MDgwMTc4ODM5MB6VDfGIbAqbukTPnwCQvo-L7U798VUWw8TEi-jfsyDEoHJgZ4UuWHedbNDjCxWhFmpYezre_R6zYiFPXXXXXXX"
			}
			]
		}
	},

sdetweil, 2022-04-12:
if u take that url, change webcal to https
and put it in a browser, does it download the ics file? 

try on browser on MM system ..

BGH-H, 2022-04-12:
it does not.

sdetweil, 2022-04-12:
it must

BGH-H, 2022-04-12:
cpoy pasted in chromium and it doesn't work. It doesn't work on a computer either

BGH-H, 2022-04-12:
Talk me through your thinking-proces

BGH-H, 2022-04-12:
p59....bla bla is redirected you too many times... try clearing cookies. 
ERRR_TOO_MANY_REDIRECTS

BGH-H, 2022-04-13:
Wow dude, that seems like the cause and a good proposed solution. Super thanks for finding that out for me. I will give it a shot.

BGH-H, 2022-04-13:
Closing....issue solved. thanks for the swift resolve.

closed 2022-04-13 by BGH-H

sdetweil, 2022-04-13:
np. it is what I originally posted 
>some of the providers are updating their access urls.. GMAIL did this